qualcomm-linux/meta-qcom #3: Resize rootfs on first boot

koenkooi, 2015-10-17:
Out of the box there is very little space (~25MiB) available, we should try to resize the partition on first boot.

The following worked for me:

```
 opkg install e2fsprogs-resize2fs ; resize2fs  /dev/disk /by-partlabel/rootfs
```

My proposal is to have a recipe called something like 'resize-rootfs.bb' that RDEPENDS on e2fsprogs-resize2fs and does 'resize2fs  /dev/disk /by-partlabel/rootfs' as on-device postinst.

ndechesne, 2015-10-19:
yes, i use resize2fs too. i wanted to find an elegant way to do that automatically.. so, we would add resize-rootfs in MACHINE_RRECOMMENDS? Did you make or are you planning to make the recipe?

koenkooi, 2015-10-19:
I'm planning on making the recipe, I'm not sure where to put it though.

ndechesne, 2015-10-19:
well, for now in recipes-bsp , in meta-qcom, it should be fine. it's not a generic recipe, since we are going to hardcode the name of the partition anyways.

ndechesne, 2016-09-06:
this can be closed, the right tools are available in meta-96boards.

closed 2016-09-06 by ndechesne