qualcomm-linux/meta-qcom #4: Network issues

koenkooi, 2015-10-17:
I'm not sure what's at fault here, but wifi won't get a proper ipv6 address:

```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP8000> mtu 1500 qdisc mq qlen 1000
    link/ether 00:0a:f5:a5:a4:d7 brd ff:ff:ff:ff:ff:ff
    inet 172.20.0.116/24 brd 172.20.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::20a:f5ff:fea5:a4d7/64 scope link 
       valid_lft forever preferred_lft forever
```

And avahi can't see anything on the network:

```
root@dragonboard-410c:~# avahi-browse -at
+  wlan0 IPv4 dragonboard-410c                              _ssh._tcp            local
+  wlan0 IPv4 dragonboard-410c [00:0a:f5:a5:a4:d7]          _workstation._tcp    local
+  wlan0 IPv4 dragonboard-410c   
root@dragonboard-410c:~#
```

Around the same time errors like these pop up:

```
[ 5775.279347] wcn36xx: ERROR SMD_EVENT (259) not supported
```

This is with the latest meta-qcom, kernel "Linux dragonboard-410c 4.2.0+linaro #1 SMP PREEMPT Sat Oct 17 14:48:34 CEST 2015 aarch64 GNU/Linux"

ndechesne, 2015-10-19:
for the 1st part, i will check. i am not sure what's going on.

for the ERROR SMD_EVENT, this can be ignored for now, we have a fix for it, i will push an update at some point

koenkooi, 2015-10-19:
If I ping the IP from my workstation I get 'no route to host' until I start pinging the workstation from the dragonboard. 
I've seen similar issues with android derived wifi drivers before, but I'm not sure what the root cause is.

ndechesne, 2015-10-19:
yes, it's a known issue:

https://www.96boards.org/forums/topic/410cubuntu-cant-ssh-in/

we are working on it...

lumag, 2023-11-04:
It is fixed now, we have new wcn36xx driver.

closed 2023-11-04 by lumag