qualcomm-linux/meta-qcom #8: fido branch

koenkooi, 2015-11-07:
I'm running into issues with repo updating git trees that have their revisions locked down in the manifest and the most failure proof fix is to create release branches everywhere. I've prepared a fido branch for meta-qcom here: https://github.com/koenkooi/meta-qcom/tree/fido

Since I'm using fido myself I also volunteer to backport things from master/jethro and keep it working with oe-core fido updates.

ndechesne, 2015-11-08:
hi,

i am fine if you want to maintain a fido branch, but mesa is too old in
fido.. and i doubt you can get freedreno to work fine with 10.4. For sure
Adreno 306 (the one we have on DB410c) isn't supported, and i am not sure
about A305 (IFC6410). Are you proposing to fix that? I am a bit worried
about exposing a 'fido' branch that is not fully functional (at least not
as much as jethro/master..). What do you think?

nico

koenkooi, 2015-11-08:
I've overlayed more recent mesa, libdrm and Xorg in meta-angstrom for v2015.06/fido, but I see you point for DISTROs not doing that.

closed 2016-04-19 by koenkooi