qualcomm-linux/meta-qcom #26: QCOM license not displayed for 32bit build

kuscsik, 2016-09-30:
Steps to reproduce:

- Call source setup-environment  and select dragonboard-410c-32.

`
kuscsik@zoli-xps:~/rpb$ source setup-environment 

The BSP for dragonboard-410c depends on packages and firmware which are covered by an End
User License Agreement (EULA). To have the right to use these binaries
in your images, you need to read and accept the following...

Would you like to read the EULA ? (y/n) y
more: stat of ../layers/meta-qcom/conf/eula/dragonboard-410c
../layers/meta-qcom/conf/eula/dragonboard-410c-32 failed: No such file or directory

Do you accept the EULA you just read? (y/n) y
EULA has been accepted.

Welcome to OpenEmbedded Reference Platform Build (OE RPB)

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

Your build environemnt has been configured with:

    MACHINE = dragonboard-410c
    SDKMACHINE = x86_64
    DISTRO = rpb-wayland

You can now run 'bitbake <target>'

Some of common targets are:
    rpb-console-image
    rpb-desktop-image
    rpb-weston-image
    core-image-base
    core-image-minimal

kuscsik@zoli-xps:~/rpb/build-rpb-wayland$ ls
`

ndechesne, 2016-09-30:
repo sync, and it will work , i just pushed a patch for that a couple of hours ago ;-)

closed 2016-09-30 by ndechesne

kuscsik, 2016-09-30:
Oh, thanks :)