qualcomm-linux/meta-qcom #28: Only one CPU with krogoth branch, massive tracebacks in dmesg

koenkooi, 2016-10-16:
I built and flashed an angstrom build yesterday and I get only one cpu:

```
root@dragonboard-410c:~# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 0

root@dragonboard-410c:~# 
```

and

```
root@dragonboard-410c:~# uname -a
Linux dragonboard-410c 4.4.23+linaro #1 SMP PREEMPT Sat Oct 15 09:52:50 CEST 2016 aarch64 aarch64 aarch64 GNU/Linux
```

Full dmesg: http://sprunge.us/UUXb

The krogoth branch at this commit:

```
commit cd38876d03bbe25e5baa623574088f67aea324e9
Author: Nicolas Dechesne <nicolas.dechesne@linaro.org>
Date:   Tue Oct 11 22:34:25 2016 +0100

    linux-linaro-qcomlt_4.4: update with Linaro 16.09 release
    
    * merged 4.4.23
    * switch to PSCI by default: so it requires the latest bootloader with PSCI
      enabled
    
    Signed-off-by: Nicolas Dechesne <nicolas.dechesne@linaro.org>
```

ndechesne, 2016-10-16:
This is very likely a firmware issue. This commit is a big change since we
are switching the platform to PSCI firmware.. So you need to update the tz
partition with the latest file from 16.09 bootloader/rescue release.

INRIX-joel-winarske, 2016-12-19:
In 16.09 there are two tz mbn:
1. tz.mbn
2. tz-psci.mbn

Are you stating we should use tz-psci.mbn instead of tz.mbn?

INRIX-joel-winarske, 2016-12-19:
Loading tz-psci.mbn instead of tz.mbn fixes the problem.  I now see four cores.

closed 2025-05-10 by github-actions[bot]