qualcomm-linux/meta-qcom #35: Enable CONFIG_NETFILTER_XT_MATCH_HASHLIMIT in kernel defconfig

kraj, 2017-04-07:
While compiling wireguard-module on dragon board 410c it fails to build due to the above option being unset

| /mnt/oe/openembedded-core/build/tmp-glibc/work/dragonboard_410c-oe-linux/wireguard-module/0.0.20170324-r0/WireGuard-0.0.20170324/src/Kbuild:30: *** "WireGuard requires CONFIG_NETFILTER_XT_MATCH_HASHLIMIT to be configured in your kernel. See https://www.wireguard.io/install/#kernel-requirements for more info".  Stop.

ndechesne, 2017-04-07:
ok, i can make that change, would that be enough:
+CONFIG_NETFILTER_XT_MATCH_HASHLIMIT=m
or does it need to be =y?
If so, i will add it to kernel/configs/distro.config in our tree

kraj, 2017-04-07:
module should be ok.

closed 2025-05-08 by github-actions[bot]