qualcomm-linux/meta-qcom #43: not able to get plug-in  on 820[gstreamer]

Madan1993, 2017-07-17:
i added gstreamer in .inc file.  i am able to get api gst-player, gstreamer-launch, gstreamer-inspect.
But not getting supported plug in.
all yocto based stuff i am using.

ndechesne, 2017-07-17:
hi, i will need a bit more detail to understand the context, your build environment and any change you have done. 

also, note that this layer is tested regularly with the OE RPB project, see https://github.com/96boards/oe-rpb-manifest.

Madan1993, 2017-07-17:
hi ndechesne,

i am trying to use gstreamer on platform snapdragaon820, using yocto project. while adding gstreamer i added IMAGE_INSTALL += "gstreamer1.0" in meta-qti-bsp/recipes-products/images/apq8096/apq8096-base-image.inc . so it build gstreamer and gets api 'gst-player', 'gstreamer-launch' etc.

i have support of gstreamer1.0
gstreamer1.0-plugins-base
gstreamer1.0-plugins-good

while playing file it get failed.
 i want to play some like .mp3, .mp4 etc. file

ndechesne, 2017-07-17:
for a 'good' gstreamer installation in an OE/Yocto based build, please include the following packages in your images:

    gstreamer1.0-plugins-bad-meta \
    gstreamer1.0-plugins-base-meta \
    gstreamer1.0-plugins-good-meta \
    gstreamer1.0-libav \

Note: none of this is specific to DB820c.

Madan1993, 2017-07-17:
under what name? i simply add, build failed
we added userparam like

inherit extrausers
EXTRA_USERS_PARAMS = "\
        useradd -P welcome david; \
        usermod -P welcome root; \
        "
is above added something like this??  if this way then what should be in place of 'EXTRA_USERS_PARAMS' ??

Madan1993, 2017-07-18:
Thanks ndechesne,
i added successfully  and built it. 
given following command
gst-play-1.0 test.mp3 

get error :
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/root/test.mp3
WARNING No decoder available for type 'application/x-id3'.
WARNING debug information: /home/le820nanosom/test_2/poky/build/tmp-glibc/work/aarch64-oe-linux/gstreamer1.0-plugins-base/1.10.2-r0/gst-plugin0
ERROR Your GStreamer installation is missing a plug-in. for file:///home/root/test.mp3
ERROR debug information: /home/le820nanosom/test_2/poky/build/tmp-glibc/work/aarch64-oe-linux/gstreamer1.0-plugins-base/1.10.2-r0/gst-plugins-:
no suitable plugins found:
/home/le820nanosom/test_2/poky/build/tmp-glibc/work/aarch64-oe-linux/gstreamer1.0-plugins-base/1.10.2-r0/gst-plugins-base-1.10.2/gst/playback/:
no suitable plugins found:
Missing decoder: ID3 tag (application/x-id3)

Reached end of play list.

Note: to play i following instruction from 
https://gstreamer.freedesktop.org/documentation/frequently-asked-questions/using.html
and also does alsa setting

closed 2017-07-18 by Madan1993