qualcomm-linux/meta-qcom #51: Bluetooth issue with newer 4.9.56 linux kernel

jayzhao93, 2017-11-27:
Hi everyone,

I built a custom yocto image for dragonboard-410c using this meta-qcom layer. Everything built fine. But after I load the image, the bluetooth is not working. Hence, when I do $ hciconfig hci0 up, it gives me this error message: Operation not supported. Btw, the wifi is working.

So, I am wondering if this is a known issue or not. So for now, i just used the older version of the linux kernel 4.9.39. This version worked perfectly.

BG
Jay

ndechesne, 2017-11-27:
hmm. right.. we've backported this patch into our kernel: https://www.spinics.net/lists/linux-arm-msm/msg30024.html, and it requires a bootloader update (this commit: https://git.linaro.org/landing-teams/working/qualcomm/lk.git/commit/?h=release/LA.BR.1.2.7-03810-8x16.0&id=372982a4d4c4922e9214ff7d6aa5348aaba602a7). 

So you probably need to update LK (e.g. aboot partition) using the latest bootloader release available on 96boards.org website.

ndechesne, 2017-11-28:
hi,

no, this is the 'linux' release that includes boot and rootfs partitions. The bootloader release contains all the binary that we flash into the other partitions (e.g. all s/w needed before we reach the boot partition). The bootloader releases are here:

http://builds.96boards.org/releases/dragonboard410c/linaro/rescue/latest/

closed 2017-11-30 by jayzhao93