qualcomm-linux/meta-qcom #52: Rocko: Build fails with

mwelchuk, 2017-11-30:
Hi,

I'm attempting to build `core-image-minimal` for `dragonboard-410c`. 
The build fails with:
```
martyn@hades:~/Documents/yocto/build-dragonboard-rocko$ bitbake core-image-minimal
Parsing recipes: 100% |############################################################################################################################################| Time: 0:00:37
Parsing of 1980 .bb files complete (0 cached, 1980 parsed). 2774 targets, 113 skipped, 0 masked, 0 errors.
NOTE: There are 1 recipes to be removed from sysroot aarch64, removing...
NOTE: There are 1 recipes to be removed from sysroot dragonboard-410c, removing...
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.36.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal-4.9"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "dragonboard-410c"
DISTRO            = "poky"
DISTRO_VERSION    = "2.4"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "rocko:f7b90ab3eaf832bd81f3efc1dab4dcf6863ac284"
meta-oe           
meta-python       
meta-multimedia   
meta-networking   = "rocko:6e3fc5b8d904d06e3aa77e9ec9968ab37a798188"
meta-qcom         = "master:0385b8d77360fcdcd9b416e5e9f0011268625d9f"

Initialising tasks: 100% |#########################################################################################################################################| Time: 0:00:03
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: core-image-minimal-1.0-r0 do_image_wic: Function failed: do_image_wic (log file is located at /home/martyn/Documents/yocto/build-dragonboard-rocko/tmp/work/dragonboard_410c-poky-linux/core-image-minimal/1.0-r0/temp/log.do_image_wic.31942)
ERROR: Logfile of failure stored in: /home/martyn/Documents/yocto/build-dragonboard-rocko/tmp/work/dragonboard_410c-poky-linux/core-image-minimal/1.0-r0/temp/log.do_image_wic.31942
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/fdisk/gptfdisk_1.0.3.bb:do_populate_sysroot', '/home/martyn/Documents/yocto/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/home/martyn/Documents/yocto/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-extended/pigz/pigz_2.3.4.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-extended/parted/parted_3.2.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/createrepo-c/createrepo-c_git.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/mtools/mtools_4.0.18.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/opkg/opkg_0.3.5.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/rpm/rpm_git.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-core/update-rc.d/update-rc.d_0.7.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/dnf/dnf_2.6.3.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/pseudo/pseudo_1.8.2.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/dosfstools/dosfstools_4.1.bb:do_populate_sysroot', '/home/martyn/Documents/yocto/meta-qcom/recipes-bsp/firmware/firmware-qcom-dragonboard410c-bootloader-sdcard_17.09.bb:do_populate_sysroot', '/home/martyn/Documents/yocto/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/martyn/Documents/yocto/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.5.bb:do_populate_sysroot', '/home/martyn/Documents/yocto/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gptfdisk-native', 'ldconfig-native', 'makedevs-native', 'depmodwrapper-cross', 'pigz-native', 'parted-native', 'prelink-native', 'createrepo-c-native', 'mtools-native', 'opkg-native', 'rpm-native', 'update-rc.d-native', 'dnf-native', 'pseudo-native', 'dosfstools-native', 'firmware-qcom-dragonboard410c-bootloader-sdcard', 'qemuwrapper-cross', 'opkg-utils-native', 'mklibs-native', 'popt-native', 'ncurses-native', 'util-linux-native', 'quilt-native', 'kmod-native', 'qemu-native', 'zlib-native', 'gnu-config-native', 'texinfo-dummy-native', 'autoconf-native', 'readline-native', 'libtool-native', 'gettext-minimal-native', 'automake-native', 'pkgconfig-native', 'xz-native', 'binutils-native', 'elfutils-native', 'expat-native', 'cmake-native', 'libxml2-native', 'file-native', 'sqlite3-native', 'openssl-native', 'curl-native', 'python3-native', 'glib-2.0-native', 'libarchive-native', 'libsolv-native', 'shadow-native', 'nss-native', 'bzip2-native', 'dbus-native', 'db-native', 'libdnf-native', 'librepo-native', 'libcomps-native', 'python3-iniparse-native', 'glibc', 'gcc-runtime', 'lzo-native', 'gtk-doc-native', 'pixman-native', 'libsdl-native', 'dtc-native', 'alsa-lib-native', 'm4-native', 'flex-native', 'bison-native', 'cryptodev-linux-native', 'makedepend-native', 'gettext-native', 'libpcre-native', 'libffi-native', 'e2fsprogs-native', 'nspr-native', 'gobject-introspection-native', 'libcheck-native', 'gpgme-native', 'attr-native', 'python3-six-native', 'linux-libc-headers', 'libgcc', 'util-macros-native', 'libpng-native', 'libxext-native', 'libxrender-native', 'libxrandr-native', 'libx11-native', 'xproto-native', 'libassuan-native', 'swig-native', 'libgpg-error-native', 'python3-setuptools-native', 'xextproto-native', 'renderproto-native', 'randrproto-native', 'libxcb-native', 'inputproto-native', 'xtrans-native', 'kbproto-native', 'unzip-native', 'xcb-proto-native', 'libxau-native', 'libpthread-stubs-native', 'libxdmcp-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_image_wic
| INFO: Creating image(s)...
| 
| ERROR: /home/martyn/Documents/yocto/meta-qcom/scripts/lib/wic/canned-wks/dragonboard410c-sd.wks:8: unrecognized arguments: --part-name
| 
| WARNING: /home/martyn/Documents/yocto/build-dragonboard-rocko/tmp/work/dragonboard_410c-poky-linux/core-image-minimal/1.0-r0/temp/run.do_image_wic.31942:1 exit 1 from 'BUILDDIR="/home/martyn/Documents/yocto/build-dragonboard-rocko" wic create "$wks" --vars "/home/martyn/Documents/yocto/build-dragonboard-rocko/tmp/sysroots/dragonboard-410c/imgdata/" -e "core-image-minimal" -o "$out/"'
| ERROR: Function failed: do_image_wic (log file is located at /home/martyn/Documents/yocto/build-dragonboard-rocko/tmp/work/dragonboard_410c-poky-linux/core-image-minimal/1.0-r0/temp/log.do_image_wic.31942)
ERROR: Task (/home/martyn/Documents/yocto/poky/meta/recipes-core/images/core-image-minimal.bb:do_image_wic) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2965 tasks of which 2958 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/martyn/Documents/yocto/poky/meta/recipes-core/images/core-image-minimal.bb:do_image_wic
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
martyn@hades:~/Documents/yocto/build-dragonboard-rocko$ 
```
What am I doing wrong?

ndechesne, 2017-11-30:
you are using meta-qcom/master branch with oe-core/rocko branch. in master branch i recently merged a patch to build the wic images, however that feature depends on a patch in oe-core which is for now only in master branch (e.g. http://cgit.openembedded.org/openembedded-core/commit/scripts/lib/wic?id=9b60e3466ed7cff0cea10815851eb1304002eb52). I am trying to get it backported into rocko.. but it's not there yet. 

You should be using meta-qcom/rocko branch instead of master, and it should work. Or you can keep using master, but then remove "wic.gz" from IMAGE_FSTYPES in the machine .conf file.

mwelchuk, 2017-11-30:
Thanks @ndechesne, literally just stumbled upon that and came here to answer my own question (should have dug that *little* further first). Cherry-picked `6f3b0a48acb22871890c4186a21d371e44751076` in poky master and all seems good.

Yes, somehow I also managed to miss that you have branches setup (I must have run `git branch` rather than `git branch -a` when I was setting up).

Sorry for the noise.

ndechesne, 2017-11-30:
np.

closed 2017-11-30 by ndechesne