qualcomm-linux/meta-qcom #55: Linux-firmware can't be installed in 'rocko' due to comflicts with firmware-qcom-dragonboard410c

koenkooi, 2017-12-22:
When using an image with 'linux-firmware' in it I get the following:

```
Collected errors:
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/a300_pfp.fw
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.b02
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.b00
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.b01
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.b03
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.mdt
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/qcom/venus-1.8/venus.b04
	But that file is already provided by package  * firmware-qcom-dragonboard410c
 * check_data_file_clashes: Package linux-firmware wants to install file /build/jenkins/v2017.12/Domoticz/machine/dragonboard-410c/build/tmp-angstrom-glibc/work/dragonboard_410c-angstrom-linux/domoticz-image/1.0-r0/rootfs/lib/firmware/a300_pm4.fw
	But that file is already provided by package  * firmware-qcom-dragonboard410c
g.
```

ndechesne, 2018-01-12:
i think this should fix this issue.

http://lists.openembedded.org/pipermail/openembedded-core/2018-January/146351.html

this is against master, and i can either try to get it backported to rocko, or i can add a bbappend in meta-qcom, otherwise.

dl9pf, 2018-02-22:
+1 same problem here.

ndechesne, 2018-02-22:
it should be fixed in master. @dl9pf which branch are you using? maybe linux-firmware has been updated in rocko, and we see the issue there too?

dl9pf, 2018-02-22:
It is rocko. Can you backport ?

ndechesne, 2018-02-22:
yes, of course. i was waiting for linux-firmware to be updated in oe-core/rocko.. sorry that I missed it. i will backport!

dl9pf, 2018-02-22:
cool, tnx.

ndechesne, 2018-02-23:
@koenkooi have you ever tried a build with my patch that went into master? looking again at , i don't think it would fix the problem..  it's still right to split out linux-firmware subpackages for the qcom firmware, but since linux-firmware itself depends on all split out packages, we still have the conflicts. or am i missing something here?

ndechesne, 2018-02-23:
I don't see any other option , and i think i need to stop putting the adreno and venus firmware in my  firmware-qcom-dragonboard410c package, and use the one from linux-firmware now.. the tricky thing will be when we have an updated release from QCOM , not yet in linux-firmware... 

do you have any other smart idea?

koenkooi, 2018-02-23:
No clever ideas apart from using RCONFLICTS + RREPLACES = linux-firmware-venus in the qcom recipes.

ndechesne, 2018-02-23:
@alimon just suggested offline to use machine specific blacklist.. i will try that. and then decide what's the best thing to do, that or simply keep the firmware from linux-firmware..

ndechesne, 2018-02-27:
Ok, i think it's fixed now. i pushed 7f06e9f1ce105b70a4bbec836626ba534f97f0f8 to rocko which includes backport of the firmware packaging in linux-firmware (using bbappend), and proper set of RREPLACES/RCONFLICTS/RPROVIDES. with that I was able to create an image with linux-firmware installed. And I could verify that the firmware packages from meta-qcom were used as higher priority.

I pushed a similar fix for master as well, b996e36d4fd0e6b155a4659381a1f65410ecd014

closed 2018-02-27 by ndechesne