qualcomm-linux/meta-qcom #61: IMAGE_FSTYPES should not use _append but ?=

dl9pf, 2018-02-22:
We should not _append to IMAGE_FSTYPES but use ?= to let the user choose the output format w/o us adding another item on-top (unconditionally). 

The current setting of IMAGE_FSTYPES_append breaks initrd builds here.

ndechesne, 2018-02-22:
you might be right.. but ... several other BSP layers do similar things meta-ti, meta-intel and even meta-yocto-bsp.. so it's not very clear what we should do..

dl9pf, 2018-02-22:
The problem is that _append will unconditionally add it to the IMAGE_FSTYPES. 
Distros will have a hard time to specify their desired output image (if it is not the wic.gz for whatever reason).
In this particular case the _append gets even added for an initrd build (which has its own IMAGE_FSTYPE = "foo" in the recipe).

IIRC the last consent is that the BSP should not set those vars. You might have an accompanying meta-qcom-example layer as in this example: https://github.com/siemens/meta-iot2000 .

dl9pf, 2018-03-07:
See https://github.com/ndechesne/meta-qcom/pull/62 .
I think it is mainly the late processing of the _append that is causing this.

I think ?= should be used as a default. 

Situation might be better if you do a IMAGE_FSTYPES +=  as this is not postprocessed and can be overridden easily by distros using your layer. We should try this.

This applies to 
`meta-qcom/conf/machine/dragonboard-410c.conf:#IMAGE_FSTYPES_append = " wic.gz"
meta-qcom/conf/machine/include/qcom-apq8016.inc:IMAGE_FSTYPES_append = " ext4.gz"
meta-qcom/conf/machine/include/qcom-apq8064.inc:IMAGE_FSTYPES_append = " ext4.gz"
meta-qcom/conf/machine/include/qcom-apq8096.inc:IMAGE_FSTYPES_append = " ext4.gz"
`

dl9pf, 2018-03-07:
To explain the issue:
In our distro configuration we set   IMAGE_FSTYPES := "${OUR_DEFAULT_IMAGE_FSTYPES}"

So we expect this to win. But _append will be processed after our redefinition happens.
Moreover the append also applies to **initrd recipes** that set their own IMAGE_FSTYPES in the recipe already and we add an unsupported IMAGE_FSTYPES there, too.

dl9pf, 2018-03-07:
Please review the attached patch.
[0001-Change-IMAGE_FSTYPES-to-not-use-_append.patch.txt](https://github.com/ndechesne/meta-qcom/files/1789856/0001-Change-IMAGE_FSTYPES-to-not-use-_append.patch.txt)

ndechesne, 2018-03-12:
hi, i've been out of office for some time. will get back to this issue at some point. i want to check the impact on our builds first.

dl9pf, 2018-03-27:
Hi !
Have you been able to test it ?

dl9pf, 2018-03-27:
Example: 
http://git.yoctoproject.org/cgit/cgit.cgi/meta-freescale/tree/conf/machine/include/imx-base.inc#n244

ndechesne, 2018-04-10:
hi, i am looking into right now.. sorry about the delay.. at first sight it looks like using ?= should work fine.. i need to check a bit more, but i might just do that..

ndechesne, 2018-04-10:
how about https://github.com/ndechesne/meta-qcom/pull/67?

ndechesne, 2018-04-10:
@alimon , care to have a look?

closed 2018-04-30 by ndechesne