qualcomm-linux/meta-qcom #65: Resolving builds.96boards.org (builds.96boards.org)... failed: Name or service not known.

dl9pf, 2018-03-07:
WARNING: firmware-qcom-dragonboard820c-01700.1-r0 do_fetch: Failed to fetch URL http://builds.96boards.org/releases/dragonboard820c/qualcomm/firmware/linux-board-support-package-r01700.1.zip, attempting MIRRORS if available
ERROR: firmware-qcom-dragonboard820c-01700.1-r0 do_fetch: Fetcher failure: Fetch command export DBUS_SESSION_BUS_ADDRESS="unix:abstract=/tmp/dbus-ahD33yNvjU,guid=840d7b839b4278f676d0f88c5a9e6815"; export SSH_AGENT_PID="7555"; export SSH_AUTH_SOCK="/home/dl9pf/.byobu/.ssh-agent"; export PATH="/home/scratch/AGL/repobuild-local/build-820c/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/usr/bin/tar-native:/home/scratch/AGL/repobuild-local/poky/scripts:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/usr/bin/aarch64-agl-linux:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot/usr/bin/crossscripts:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/usr/sbin:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/usr/bin:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/sbin:/home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/recipe-sysroot-native/bin:/home/scratch/AGL/repobuild-local/poky/bitbake/bin:/home/scratch/AGL/repobuild-local/build-820c/tmp/hosttools"; export HOME="/home/dl9pf"; /usr/bin/env wget -t 2 -T 30 -nv --passive-ftp --no-check-certificate -P /home/dl9pf/.yocto/shared_downloads/ 'http://builds.96boards.org/releases/dragonboard820c/qualcomm/firmware/linux-board-support-package-r01700.1.zip' --progress=dot -v failed with exit code 4, output:
--2018-03-07 18:13:13--  http://builds.96boards.org/releases/dragonboard820c/qualcomm/firmware/linux-board-support-package-r01700.1.zip
Resolving builds.96boards.org (builds.96boards.org)... failed: Name or service not known.
wget: unable to resolve host address ‘builds.96boards.org’

ERROR: firmware-qcom-dragonboard820c-01700.1-r0 do_fetch: Fetcher failure for URL: 'http://builds.96boards.org/releases/dragonboard820c/qualcomm/firmware/linux-board-support-package-r01700.1.zip'. Unable to fetch URL from any source.
ERROR: firmware-qcom-dragonboard820c-01700.1-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/scratch/AGL/repobuild-local/build-820c/tmp/work/dragonboard_820c-agl-linux/firmware-qcom-dragonboard820c/01700.1-r0/temp/log.do_fetch.13797
ERROR: Task (/home/scratch/AGL/repobuild-local/meta-qcom/recipes-bsp/firmware/firmware-qcom-dragonboard820c_01700.1.bb:do_fetch) failed with exit code '1'

ndechesne, 2018-03-08:
I think it should be fixed now. We had some dns issue on our side. Can you double check?

dl9pf, 2018-03-08:
Yep, works we can close. Tnx!

closed 2018-03-08 by dl9pf