qualcomm-linux/meta-qcom #75: Rocko branch bbappends wrong version

dl9pf, 2018-07-29:
You bbappend gpsd-3.17 whereas http://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-navigation/gpsd?h=rocko&id=352531015014d1957d6444d114f4451e241c4d23 has 3.16 . 

Error: 
Parsing recipes: 100% |################################################################################################################################################################| Time: 0:00:57
Parsing of 2367 .bb files complete (0 cached, 2367 parsed). 3262 targets, 297 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:                                                                                                                                                                     
  /build-master2/meta-qcom/openembedded-layer/recipes-navigation/gpsd/gpsd_3.17.bbappend

ndechesne, 2018-07-30:
tl;tr
i think it should be fixed now. 

I often use meta-qcom along the side with meta-backports which we maintain on git.linaro.org. We bring a bunch of 'newer' recipes in meta-backports that we need for our builds, and we brought gpsd 3.17 there (for rocko). I have now added a mechanism to let me add patches in meta-qcom that are used/applied only when meta-backports is part of the build.

thanks for reporting this issue.

lumag, 2023-11-04:
rocko is no longer supported

commit 9ec24488 (2018-07-30): gpsd: move bbappend in -backports

This GPSd patch applies to GPSd 3.17 which is brought (in rocko) by
meta-backports only. meta-qcom users who do not use meta-backports would see a
parse warning or issue, since 3.17 is not available in rocko.

Fixes #75.
closed 2023-11-04 by lumag