qualcomm-linux/meta-qcom #95: thud branch ath10k fix does not build properly with yocto

bbrother, 2019-02-25:
The ath10k fix from pull request #93 breaks my build with yocto. It seems like the arch specific pre install function messes up the noarch linux-firmware recipe.  I get the follwing bitbake error:
```
No match for argument: linux-firmware-ath10k
Error: Unable to find a match
```

ndechesne, 2019-02-25:
ouch. @alimon can you please check? let me know if you think we should revert..

bbrother, 2019-02-25:
Note, I had to force it to try to install ath10k the MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS inclusion of it failed quietly and I only noticed by looking at the rootfs manifest that it was missing.

It does work if I remove dragonboard-820c_ from the do_install_prepend function.

lumag, 2023-11-04:
The thud branch is no longer supported.

closed 2023-11-04 by lumag