qualcomm-linux/meta-qcom #394: make-mod-scripts is not buildable with linux-linaro-qcomlt

daniel-thompson, 2021-11-25:
If we enable any feature or package that requires out-of-tree kernel modules to be built then meta-qcom based builds fail because linux-linaro-qcomlt is not compatible with make-mod-scripts. In particular this makes is difficult to set the tools-profile image feature (because it tries to build lttng-modules).

So far I have tested only the dunfell packages. Is linux-linaro-qcomlt expected to support out-of-tree kernel modules?

Currently I have adopted the following trivial workaround but if this is something that should be fixed...

~~~
# Disable LTTng dependencies when building for qcom family parts. This is
# required because linux-linaro-qcomlt does not support out-of-tree
# module builds so we must avoid pulling in lttng-modules.
LTTNGUST:pn-packagegroup-core-tools-profile:qcom = ""
LTTNGTOOLS:pn-packagegroup-core-tools-profile:qcom = ""
LTTNGMODULES:pn-packagegroup-core-tools-profile:qcom = ""
~~~

ndechesne, 2021-11-25:
well, I would say that it is not expected to fail.. what's the simplest way to reproduce the issue?

daniel-thompson, 2021-11-25:
The simplest way is just to `bitbake make-mod-scripts` (normally the recipe gets pulled by the following line in module-base.bbclass, `do_configure[depends] += "make-mod-scripts:do_compile"`).

An example error log is below, it is based on dunfell/db410c (using poky + meta-qcom):
~~~
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/bison/bison_3.5.4.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-kernel/kmod/kmod-native_git.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-extended/bc/bc_1.07.1.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/binutils/binutils-cross_2.34.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-connectivity/openssl/openssl_1.1.1l.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/gcc/gcc-cross_9.3.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-support/gmp/gmp_6.2.0.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-support/lzop/lzop_1.04.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-core/glibc/glibc_2.31.bb:do_populate_sysroot', '/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/gcc/gcc-runtime_9.3.bb:do_populate_sysroot', 'virtual:native:/build/drt/openembedded/poky-3.1-dunfell/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: ['kmod-native', 'openssl-native', 'lzop-native', 'python3-native', 'gtk-doc-native', 'lzo-native', 'libnsl2-native', 'gdbm-native', 'util-linux-native', 'bzip2-native', 'sqlite3-native', 'libtirpc-native', 'libffi-native', 'libcap-ng-native', 'libpcre2-native']
| NOTE: Skipping as already exists in sysroot: ['bison-native', 'pkgconfig-native', 'bc-native', 'binutils-cross-aarch64', 'gcc-cross-aarch64', 'pseudo-native', 'gmp-native', 'quilt-native', 'glibc', 'gcc-runtime', 'patch-native', 'automake-native', 'libtool-native', 'xz-native', 'gnu-config-native', 'flex-native', 'texinfo-dummy-native', 'gettext-minimal-native', 'autoconf-native', 'zlib-native', 'readline-native', 'libmpc-native', 'linux-libc-headers', 'mpfr-native', 'libgcc', 'attr-native', 'm4-native', 'ncurses-native']
| DEBUG: sed -e 's:^[^/]*/:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native/:g' /build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/sysroots-components/x86_64/openssl-native/fixmepath /build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/sysroots-components/x86_64/python3-native/fixmepath /build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/sysroots-components/x86_64/gtk-doc-native/fixmepath /build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/sysroots-components/x86_64/libpcre2-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/pkgdata/dragonboard-410c:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/pseudo/:g' -e 's:FIXME_LOGFIFO:/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/temp/fifo.206557:g'
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| NOTE: make HOSTCC=gcc  -isystem/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native/usr/lib                         -L/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/build/drt/openembedded/poky- [... cut ...]
| make: Leaving directory '/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work-shared/dragonboard-410c/kernel-source'
| ERROR: oe_runmake failed
| WARNING: /build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/temp/run.do_configure.206557:1 exit 1 from 'exit 1'
| ERROR: Execution of '/build/drt/openembedded/poky-3.1-dunfell/build-dragonboard-410c/tmp/work/dragonboard_410c-poky-linux/make-mod-scripts/1.0-r0/temp/run.do_configure.206557' failed with exit code 1
~~

lumag, 2021-11-25:
@daniel-thompson I could not reproduce `bitbake make-mod-scripts` failure this either with master or with dunfell branch, either with rpb distro or with nodistro (and using jut oe-core, meta-oe and meta-qcom). Could you please clarify, which kernel recipe is building the shared kernel source (and what is the commit id).

ndechesne, 2021-11-25:
thanks. I could reproduce with dunfell.

We need to cherry pick 
ddad8183490c (make-mod-scripts: pass CROSS_COMPILE to configure and build)

in oe-core/dunfell, it was merged in oe-core/master. I suspect it was needed in 'recent' kernel version.

ndechesne, 2021-11-25:
that seems to be the context: https://lists.openembedded.org/g/openembedded-core/message/150143. Seems it's needed for 5.12++

I will ask the dunfell maintainer to take this patch.

lumag, 2023-11-04:
Resolved in oe-core

closed 2023-11-04 by lumag