qualcomm-linux/meta-qcom #490: Trying to find MDM9607 support

Sergnone, 2022-11-09:
Hello everyone. I try to compile the basic template FW for MDM9607 CPU.
But system says: "MACHINE=mdm9607 is invalid. Please set a valid MACHINE in your local.conf"
I figured out that there are no such machine in meta-qcom, but this was mentioned some time ago as existing:
https://github.com/the-modem-distro/pinephone_modem_sdk/issues/1

The global question, that I have. How can I adjust properly my environment specially for MDM9607 CPU?? Due I cannot find any mentions in meta-qcom. 

Thanks.

ndechesne, 2022-11-09:
msm9607 is not supported by meta-qcom, there is no machine meta data defined for this device, and I am not sure what the upstream kernel support really is for this platform.

closed 2022-11-09 by ndechesne

lumag, 2022-11-09:
@Sergnone you can probably start by copying & modifying the sdx55.conf. However I agree with @ndechesne , the kernel might require additional patches. You can ask @superna9999 (narmstrong on IRC), he was recently sending patches for this platform.

And, of course, patches for meta-qcom are also welcomed.

superna9999, 2022-11-09:
@superna9999 indeed it seems upstreaming for mdm9607 stopped in the middle, there's some bindings and driver changes but DT is missing