qualcomm-linux/meta-qcom #616: info: flash core-image-minimal into rb3gen2 qcs6490

Scott31393, 2024-07-23:
Hi All,

Sorry for bothering. I'm working on top of your meta-qcom

My target is to run mainline linux kernel into rb3gen2 (6490) board. I want ask if you can share some guide to flash the yocto images generated with this layer. I'm able to build/flash the qcom bsp based on the following yocto manifest: https://github.com/quic-yocto/qcom-manifest/blob/qcom-linux-kirkstone/qcom-6.6.28-QLI.1.1-Ver.1.1.xml using qdl tool.

My plan is to do a test on mainline linux kernel. I think to flash those images into the board you are using a different partitioning schema with a different approach. When I'm trying to flash the bsp using the following cmd I'm getting:

```
fastboot flash boot ./boot-qcs6490-rb3gen2-qcom-armv8a.img
Sending 'boot' (8264 KB)                           OKAY [  0.444s]
Writing 'boot'                                     FAILED (remote: 'No such partition.')
fastboot: error: Command failed
```
Do you have some tips/guide to share? :)
Many thanks in advance!

Thanks & Regards,
Tommaso

Scott31393, 2024-07-23:
Update:

Fixing the following issue from: recipes-kernel/images/linux-qcom-uki.bb

```
NOTE: Resolving any missing task queue dependencies
ERROR: /qualcom/yocto-mainline/poky/meta-qcom/recipes-kernel/images/linux-qcom-uki.bb:do_configure[depends], dependency ${@ in '      systemd-boot:do_deploy     virtual/kernel:do_deploy      ${@ '${INITRAMFS_IMAGE}:do_image_complete' if d.getVar('INITRAMFS_IMAGE') else ''}' does not contain exactly one ':' character.
 Task 'depends' should be specified in the form 'packagename:task'
ERROR: Command execution failed: Exited with 1
```
I'm able to **bitbake esp-qcom-image** using that image I can flash efi partition given by the vendor bsp using:

```
fastboot flash efi ./boot-qcs6490-rb3gen2-qcom-armv8a.img
```

Unfortunately now I'm getting some issue from the kernel that has no bootargs.
I guess this because **KERNEL_CMDLINE_EXTRA** variable is empty and the kernel is not able to mount the fs.

```
Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Linux version 6.6.35-yocto-standard (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 14.1.0, GNU ld (GNU Binutils) 2.42.0.20240620) #1 SMP PREEMPT Fri Jun 28 17:07:28 UTC 2024
Machine model: Qualcomm Technologies, Inc. Robotics RB3gen2 addons vision mezz platform
efi: EFI v2.7 by Qualcomm Technologies, Inc.
efi: MEMATTR=0x9ccb3698 RNG=0x9cc54018 MEMRESERVE=0x9cc5ca18 
random: crng init done
Reserved memory: created CMA memory pool at 0x00000000ff000000, size 12 MiB
OF: reserved mem: initialized node adsp-heap, compatible id shared-dma-pool
OF: reserved mem: 0x00000000ff000000..0x00000000ffbfffff (12288 KiB) map reusable adsp-heap
Reserved memory: created CMA memory pool at 0x00000000fd000000, size 32 MiB
OF: reserved mem: initialized node secure-cdsp-region, compatible id shared-dma-pool
OF: reserved mem: 0x00000000fd000000..0x00000000feffffff (32768 KiB) map reusable secure-cdsp-region
Reserved memory: created CMA memory pool at 0x00000000fc000000, size 16 MiB
OF: reserved mem: initialized node qcom,audio-ml, compatible id shared-dma-pool
OF: reserved mem: 0x00000000fc000000..0x00000000fcffffff (16384 KiB) map reusable qcom,audio-ml
Reserved memory: created CMA memory pool at 0x00000000fac00000, size 20 MiB
OF: reserved mem: initialized node qcom,qseecom, compatible id shared-dma-pool
OF: reserved mem: 0x00000000fac00000..0x00000000fbffffff (20480 KiB) map reusable qcom,qseecom
Reserved memory: created CMA memory pool at 0x00000000f9c00000, size 16 MiB
OF: reserved mem: initialized node qcom,qseecom-ta, compatible id shared-dma-pool
OF: reserved mem: 0x00000000f9c00000..0x00000000fabfffff (16384 KiB) map reusable qcom,qseecom-ta
Reserved memory: created CMA memory pool at 0x00000000f7000000, size 44 MiB
OF: reserved mem: initialized node mem-dump-region, compatible id shared-dma-pool
OF: reserved mem: 0x00000000f7000000..0x00000000f9bfffff (45056 KiB) map reusable mem-dump-region
OF: reserved mem: 0x0000000080000000..0x00000000805fffff (6144 KiB) nomap non-reusable memory@80000000
OF: reserved mem: 0x0000000080700000..0x00000000807fffff (1024 KiB) nomap non-reusable memory@80600000
OF: reserved mem: 0x0000000080800000..0x000000008085ffff (384 KiB) nomap non-reusable memory@80800000
OF: reserved mem: 0x0000000080860000..0x000000008087ffff (128 KiB) nomap non-reusable memory@80860000
OF: reserved mem: 0x0000000080884000..0x0000000080893fff (64 KiB) nomap non-reusable memory@80880000
OF: reserved mem: 0x00000000808ff000..0x00000000808fffff (4 KiB) nomap non-reusable memory@808ff000
OF: reserved mem: 0x0000000080900000..0x0000000080afffff (2048 KiB) nomap non-reusable memory@80900000
OF: reserved mem: 0x0000000080b00000..0x0000000080bfffff (1024 KiB) nomap non-reusable memory@80b00000
OF: reserved mem: 0x0000000080c00000..0x00000000817fffff (12288 KiB) nomap non-reusable memory@80c00000
OF: reserved mem: 0x0000000084300000..0x00000000847fffff (5120 KiB) nomap non-reusable camera@84300000
OF: reserved mem: 0x0000000084800000..0x00000000860fffff (25600 KiB) nomap non-reusable wpss@84800000
OF: reserved mem: 0x0000000086100000..0x00000000888fffff (40960 KiB) nomap non-reusable adsp@86100000
OF: reserved mem: 0x0000000088900000..0x000000008a6fffff (30720 KiB) nomap non-reusable cdsp@88900000
OF: reserved mem: 0x000000008a700000..0x000000008adfffff (7168 KiB) nomap non-reusable memory@8b200000
OF: reserved mem: 0x000000008ae00000..0x000000008b2fffff (5120 KiB) nomap non-reusable cvp@8ae00000
OF: reserved mem: 0x000000008b300000..0x000000008b30ffff (64 KiB) nomap non-reusable memory@8b700000
OF: reserved mem: 0x000000008b310000..0x000000008b319fff (40 KiB) nomap non-reusable ipa-gsi@8b310000
OF: reserved mem: 0x000000008b31a000..0x000000008b31bfff (8 KiB) nomap non-reusable gpu-microcode@8b31a000
OF: reserved mem: 0x000000008b800000..0x000000009adfffff (251904 KiB) nomap non-reusable mpss@8b800000
OF: reserved mem: 0x00000000c0000000..0x00000000c00fffff (1024 KiB) nomap non-reusable tz-stat@c0000000
OF: reserved mem: 0x00000000c0100000..0x00000000c12fffff (18432 KiB) nomap non-reusable tags@c0100000
OF: reserved mem: 0x00000000c1300000..0x00000000c17fffff (5120 KiB) nomap non-reusable qtee@c1300000
OF: reserved mem: 0x00000000c1800000..0x00000000c33fffff (28672 KiB) nomap non-reusable trusted_apps@c1800000
OF: reserved mem: 0x00000000d0600000..0x00000000d06fffff (1024 KiB) nomap non-reusable debug-vm@d0600000
Zone ranges:
  DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[... 809 lines cut ...]
[... cut ...]
 show_stack+0x18/0x24
 dump_stack_lvl+0x48/0x60
 dump_stack+0x18/0x24
 panic+0x304/0x35c
 mount_root_generic+0x280/0x350
 mount_root+0x178/0x33c
 prepare_namespace+0x6c/0x2b0
 kernel_init_freeable+0x2e8/0x3e4
 kernel_init+0x28/0x1dc
 ret_from_fork+0x10/0x20
SMP: stopping secondary CPUs
Kernel Offset: disabled
CPU features: 0x0,c0000000,70020143,1000720b
Memory Limit: none
---[ end Kernel panic - not syncing: Invalid configuration from end user prevents continuing ]---

```

I will continue tomorrow :)

Scott31393, 2024-07-23:
Hi All,

Finally I'm able to run the core image minimal, below some steps. First off all I flash the qcom yocto image into the the **qcs6490rb3gen2** board, or flash the prebuild images.

Then:

```
# step 1: build required yocto images

git clone git://git.yoctoproject.org/poky
git clone https://github.com/openembedded/meta-openembedded.git
cd poky
source oe-init-build-env ../build-qcom-armv8a

bitbake-layers add-layer ../meta-qcom
bitbake-layers add-layer ../meta-openembedded/

bitbake initramfs-qcom-image core-image-minimal esp-qcom-image
------------------------------------------------------------------------------------------

# step 2: create uki.efi image
# ukify tool is located at: tmp/sysroots-components/x86_64/systemd-boot-native/usr/bin/ukify

ukify build --efi-arch=aa64  \
	--stub=./linuxaa64.efi.stub \
	--linux=./Image \
	--initrd=./initramfs-qcom-image-qcom-armv8a.cpio.gz \
	--cmdline="root=/dev/disk/by-partlabel/system rw rootwait console=ttyMSM0,115200n8 \
				earlycon pcie_pme=nomsi kernel.sched_pelt_multiplier=4 rcupdate.rcu_expedited=1 rcu_nocbs=0-7 k \
				pti=off kasan=off kasan.stacktrace=off no-steal-acc page_owner=on swiotlb=128" \
	--devicetree=./qcs6490-rb3gen2.dtb \
	--output=./uki.efi

# copy uki.efi into esp-qcom-image-qcom-armv8a.rootfs.vfat image (efi.img)

sudo mount esp-qcom-image-qcom-armv8a.rootfs.vfat /mnt
sudo cp uki.efi /mnt/EFI/Linux/linux-qcom-armv8a.efi
sudo umount /mnt
------------------------------------------------------------------------------------------

# step 3: flash efi and system partition

adb shell reboot bootloader
fastboot erase efi
fastboot erase system
fastboot flash efi ./esp-qcom-image-qcom-armv8a.rootfs.vfat
fastboot flash system ./core-image-minimal-qcom-armv8a.rootfs-20240723215932.ext4
fastboot reboot
```

core-image-minimal logs:
```
Poky (Yocto Project Reference Distro) 5.0+snapshot-e18d60deb0496f7c91f2de900d6c024b45b7910a qcom-armv8a /dev/ttyMSM0
qcom-armv8a login: root
root@qcom-armv8a:~# uname -a
Linux qcom-armv8a 6.6.35-yocto-standard #1 SMP PREEMPT Fri Jun 28 17:07:28 UTC 2024 aarch64 GNU/Linux
root@qcom-armv8a:~# platform c440000.spmi:pmic@2:temp-alarm@2400: deferred probe pendi
```
Question: 
 - is there some supported image with hdmi and wayland backend enabled?

Thanks in advance,
Tommaso

lumag, 2024-07-25:
@Scott31393 For uki image you should be able to run `bitbake linux-qcom-uki`. If that doesn't work, we have an issue that must be fixed.

Regarding HDMI. Unfortunately the HDMI support wasn't backported yet, you can try combining the logic in #593 and the patch at https://lore.kernel.org/linux-arm-msm/20240528141954.7567-1-quic_vvalluru@quicinc.com/

lumag, 2024-07-25:
Regarding wayland. This question is off-topic for the meta-qcom (it is purely a BSP and doesn't provide any higher-level integration). As a part of the [RPB effort](https://github.com/96boards/oe-rpb-manifest) we have been providing the RPB-weston [image](https://github.com/96boards/meta-rpb/blob/master/recipes-samples/images/rpb-weston-image.bb), but this image hasn't been tested for quite a while.

Scott31393, 2024-07-25:
Hi @lumag,

> @Scott31393 For uki image you should be able to run `bitbake linux-qcom-uki`. If that doesn't work, we have an issue that must be fixed.
> 
I will work on fix that, thanks.

> Regarding HDMI. Unfortunately the HDMI support wasn't backported yet, you can try combining the logic in #593 and the patch at https://lore.kernel.org/linux-arm-msm/20240528141954.7567-1-quic_vvalluru@quicinc.com/
Thanks for sharing. Will do a try on this.

What about a new machine **qcs6490.conf** or **rb3gen2.conf** that inherit from generic **qcom-armv8a.conf** and put here some board specific customization like: **KERNEL_CMDLINE_EXTRA** and others? Do you agree on that or I'm completely wrong?

Thanks,
Tommaso

lumag, 2024-07-25:
The meta-qcom idea is to have a generic machines only. The rootfs should work on every supported board without the need to further customize it (such customizations often end up being poorly designed or not really fitting). 
You can still put custom kernel command line options to `qcom-armv8a.conf` (or to `local.conf`), they are specied per the dtb.

```
KERNEL_CMDLINE_EXTRA[sdm845-db845c] ?= "clk_ignore_unused pd_ignore_unused"
```

Scott31393, 2024-07-25:
> The meta-qcom idea is to have a generic machines only. The rootfs should work on every supported board without the need to further customize it (such customizations often end up being poorly [...]
Already saw that, thanks the same.
Thanks for sharing the idea behind.

Regards,
Tommaso

Thejesh99, 2024-07-30:
https://github.com/Linaro/meta-qcom/issues/616#issuecomment-2246432930 can anyone provide a more detailed yocto build for qualcomm robotics RB3 gen 2 vision kit

ricardosalveti, 2024-07-31:
> [#616 (comment)](https://github.com/Linaro/meta-qcom/issues/616#issuecomment-2246432930) can anyone provide a more detailed yocto build for qualcomm robotics RB3 gen 2 vision kit

There is the downstream BSP which was released earlier this month (https://docs.qualcomm.com/bundle/publicresource/topics/80-70014-115/overview.html?product=1601111740013072 and https://github.com/quic-yocto/meta-qcom-hwe / https://github.com/quic-yocto/qcom-manifest). BSP is kirkstone based at the moment (work ongoing to have an updated version).

Thejesh99, 2024-07-31:
can anyone give me the correct machine name for robotics rb3 gen 2

Thejesh99, 2024-07-31:
how to check pre build images in the qualcomm robotics rb3 gen 3

ricardosalveti, 2024-07-31:
> can anyone give me the correct machine name for robotics rb3 gen 2

qcm6490 on the downstream BSP.

ricardosalveti, 2024-07-31:
> how to check pre build images in the qualcomm robotics rb3 gen 3

For the BSP:

https://artifacts.codelinaro.org/artifactory/qli-ci/flashable-binaries/qimpsdk/qcm6490/x86/qcom-6.6.28-QLI.1.1-Ver.1.1_qim-product-sdk-1.1.3.zip -> qcm6490/qcom-multimedia-image.

As described at https://docs.qualcomm.com/bundle/publicresource/topics/80-70014-253/ubuntu_host.html?product=1601111740013072

Thejesh99, 2024-08-01:
I'm encountering an issue where no login page is shown while following the steps described at [Qualcomm Documentation](https://docs.qualcomm.com/bundle/publicresource/topics/80-70014-253/ubuntu_host.html?product=1601111740013072).

How can I flash the prebuilt images, and where can I find prebuilt images for the Qualcomm Robotics RB3 Gen 2 Vision Kit?
[image]

ricardosalveti, 2024-08-01:
> I'm encountering an issue where no login page is shown while following the steps described at [Qualcomm Documentation](https://docs.qualcomm.com/bundle/publicresource/topics/80-70014-253/ubuntu_host.html?product=1601111740013072).

Would you mind sending your question at https://mysupport.qualcomm.com/supportforums/s/topic/0TO4V000000Ey0jWAC/qualcomm-linux? That would be the proper location for debugging issues with qualcomm linux itself.

Thejesh99, 2024-08-06:
how to remove flash from the qualcomm robotics rb 3 gen 3

obbardc, 2025-04-09:
I think this issue can be closed.

closed 2025-04-09 by ndechesne