qualcomm-linux/meta-qcom #680: Define a co-maintainership with Qualcomm, and migrate the layer to a Qualcomm hosted Github org

ndechesne, 2024-11-22:
hey,

as discussed with a few people privately, we intend to migrate meta-qcom into a new Github org (most likely github.com/quic-yocto) in the near future. Qualcomm has now started to develop a BSP based on Yocto development branch, using linux-yocto kernel (e.g. meta-qcom-hwe layer), and there is no strong reason to maintain 2 competing Yocto BSP.  We want to merge meta-qcom and meta-qcom-hwe 'master' branches and have all development done jointly on meta-qcom. 

We have a rough goal for this migration to be done between Dec 24 and Feb 25. 

There are many technical details to iron out, so let's use this issue to coordinate. 

Here is a summary of what we've discussed so far

1. The migration should be implemented as a Github ownership transfer, with redirection, to minimize the chance of breaking users
2. meta-qcom is a BSP layer, and any policy/distro configs should be done outside of meta-qcom. Qualcomm maintains meta-qcom-distro for that purpose
3. We want to migrate the current master branch (and future yocto stable branch will follow from that), but we are not doing any retro-active changes. meta-qcom-hwe will continue to support QC Linux 1.x releases with the kirkstone branch, and all other branches on meta-qcom won't change.
4. CI: we will move away from current Linaro tuxmake based CI for master branch and use existing Meta-qcom-hwe (main) CI infrastructure.
5. CI: Keep Linaro CI for old branches (through tuxmake)
6. Machines. Meta-qcom has ‘generic’ machines only, and no plans to add board specific machines. Linaro will continue to maintain/support the generic machines in the merged layer. There are discussions/efforts to support genericarm64 and genericarm machines from meta-yocto-bsp, but for that this is in addition to meta-qcom generic machines. Qualcomm is interested in the generic machines from meta-yocto-bsp, in addition to the specific machine conf file currently in meta-qcom-hwe (which will migrate to meta-qcom)
7. Meta-qcom: find . -name *.bb | wc -l  108, let’s review them and remove unnecessary recipes (If any)
8. Understand what initramfs-firmware-image.bb and firmware-woa  are, and how we want to support them after the merge.
9. Meta-qcom has support for meta-linux-mainline kernel recipes. is anyone using that? do we want to maintain them?
10. Meta-qcom maintains Mesa_git recipe. Is that still needed? For which purpose? 
11. What are the Binary blobs which are needed in meta-qcom, and how do we handle their redistribution.
12. meta-qcom will be open for contributions, Qualcomm and Linaro will co-maintain and thus review external contributions.
13. meta-qcom-hwe supports 2 flavor of BSP (Base and Custom), and these 2 profiles will continue to exist. 
14. Future Qualcomm Linux releases will be based off meta-qcom only (e.g. 2.x, 3.x..), and will be maintain by Qualcomm for productization/commercialization. 

We will shortly open a branch (somewhere still TBD) on github.com/quic-yocto, to start working on this

lumag, 2024-11-22:
Here is a list of all the recipes and
appends in the meta-qcom layer together with their descriptions and
some notes.

./recipes-bsp/hexagon-dsp-binaries/hexagon-dsp-binaries_20241017.bb

   fastrpc_shell and libraries for the DSP, redistributable, covered by
   LICENSE.qcom.txt

./recipes-bsp/firmware/firmware-ath6kl_git.bb

    additional ath6k firmware, not suitable for linux-firmware, redistributable

./recipes-bsp/firmware/firmware-qcom-dragonboard410c-bootloader-sdcard_17.09.bb
./recipes-bsp/firmware/firmware-qcom-dragonboard845c_20190529180356-v4.bb
./recipes-bsp/firmware/firmware-qcom-rb3gen2_00039.2.bb

    recipes to package redistributable bootloader and firmware binaries for
    supported devices.

    Note: DB845c archive and recipe also install firmware for the Renesas xHCI
    PCI controller. It has been provided to us by Thundercomm in the same ZIP
    archive and claimed to be covered by LICENSE.qcom.txt, but further
    investigation could not prove this claim. It is being used by the layer on
    the premises of being abandonware by the Renesas side, but we don't have a
    good enough licence to get it into the linux-firmware. Maybe Qualcomm can
    help with that by providing a bigger push on Renesas, especially granted
    that RB3gen2 also uses the same chip (which actually requires the
    firmware).

./recipes-bsp/firmware/firmware-qcom-dragonboard-apq8074.bb
./recipes-bsp/firmware/firmware-qcom-ifc6410.bb
./recipes-bsp/firmware/firmware-qcom-ifc6560.bb
./recipes-bsp/firmware/firmware-qcom-sm8150-hdk.bb
./recipes-bsp/firmware/firmware-qcom-sm8350-hdk.bb
./recipes-bsp/firmware/firmware-qcom-sm8450-hdk.bb
./recipes-bsp/firmware/firmware-qcom-sm8550-hdk.bb
./recipes-bsp/firmware/firmware-qcom-sm8650-hdk.bb

    recipes to package non-redistributable binary firmware for supported
    devices (HDKs, devboards). Requires the user to explicitly specify the
    NON-HLOS.bin / adreno.tar.gz / dspso.bin (pending) in order to generate
    non-empty packages.

./recipes-bsp/firmware-nexus/firmware-qcom-nexus4.bb
./recipes-bsp/firmware-nexus/firmware-qcom-nexus5.bb
./recipes-bsp/firmware-nexus/firmware-qcom-nexus5x.bb
./recipes-bsp/firmware-nexus/firmware-qcom-nexus6.bb
./recipes-bsp/firmware-nexus/firmware-qcom-nexus6p.bb
./recipes-bsp/firmware-nexus/firmware-qcom-nexus7-2013.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel2.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel3.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel3a.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel4.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel4a-5g.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel4a.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel5.bb
./recipes-bsp/firmware-nexus/firmware-qcom-pixel5a-5g.bb

     recipes to package non-redistributable binary firmware for supported
     Google phones and tablets. Downloads firmware automatically from the
     Google's website.

./recipes-bsp/packagegroups/packagegroup-firmware-dragonboard-apq8074.bb
./recipes-bsp/packagegroups/packagegroup-firmware-dragonboard410c.bb
./recipes-bsp/packagegroups/packagegroup-firmware-dragonboard820c.bb
./recipes-bsp/packagegroups/packagegroup-firmware-dragonboard845c.bb
./recipes-bsp/packagegroups/packagegroup-firmware-ifc6410.bb
./recipes-bsp/packagegroups/packagegroup-firmware-ifc6560.bb
[... 156 lines cut ...]
./recipes-graphics/xorg-xserver/xserver-xorg_%.bbappend
./recipes-graphics/xorg-xserver/xserver-xf86-config_0.1.bbappend

    Check if these can be dropped. Might be handled in the upstream

./recipes-graphics/mesa/mesa.bbappend

    Platform-specific customization to enable freedreno driver

./recipes-kernel/linux/linux-yocto_6.6.bbappend
./recipes-kernel/linux/linux-yocto_%.bbappend

    Platform-specific linux-yocto implementation

./dynamic-layers/openembedded-layer/recipes-devtools/android-tools/android-tools-conf-configfs_%.bbappend

    Platform-specific ADBD customization

./dynamic-layers/openembedded-layer/recipes-navigation/gpsd/gpsd_%.bbappend

    Qualcomm-specific GPSd driver, works only on db410c

--
With best wishes
Dmitry

ricardosalveti, 2024-11-22:
> * What are the Binary blobs which are needed in meta-qcom, and how do we handle their redistribution.

The non-redistributable binaries would need to be removed (including renesas xhci), should we create a new repository under Linaro for them (e.g. meta-qcom-linaro)?

lumag, 2024-11-22:
I think meta-qcom-linaro makes sense. At the same time ideally Qualcomm should try utilizing their power to get a sane licence for that file because, if I remember correctly, it is also **required** to get that host to work on RB3gen2 (old RB3 had it flashed in the ROM, I don't know if RB3gen2 has the ROM or not).

ndechesne, 2024-11-25:
@lumag can you please confirm if the Renesas firmware you have (for the original RB3) is the same as here: https://www.renesas.com/en/products/interface/usb-switches-hubs/upd720201-usb-30-host-controller#documents? That's the one which is recommended to use for RB3 Gen2 (see https://docs.qualcomm.com/bundle/publicresource/topics/80-70015-254/how_to.html#update-usb-and-ethernet-controller-firmware). It is not redistributed by QCOM, but there are instructions for users to download it and flash it.

ndechesne, 2024-11-25:
At least based on RB3 and RB3 Gen2 schematics, it's indeed the same Renesas bridge being used on both (uPD720201K8-701-BAC-A)

lumag, 2024-11-25:
Yes, it's the same.

My point is that if we distribute rootfs images, they (ideally) should just work without downloading any binaries. Likewise if a user builds rootfs, it should be possible to provide the rom / zip file at the build time. I'm quite insisting here because if a company develops an end-user product using RB3gen2, they should be able to support user products (like providing rootfs, providing updates, etc), without users having to download anything for their device from the 3rd party vendor (Renesas).

ricardosalveti, 2024-11-25:
From our investigation there is no ROM in rb3gen2 and we are bound to Renesas terms. It is indeed a pain to manage.

lumag, 2024-11-26:
Regarding meta-qcom-linaro. I think instead I will spawn linux-msm/meta-qcom-extras and start moving questionable recipes to that repo. @davidinux @ndechesne @ricardosalveti WDYT?

ndechesne, 2024-11-26:
> Regarding meta-qcom-linaro. I think instead I will spawn linux-msm/meta-qcom-extras and start moving questionable recipes to that repo. @davidinux @ndechesne @ricardosalveti WDYT?

This is a good idea. I was not sure meta-qcom-linaro was great.. I like your idea much better.

ricardosalveti, 2024-11-26:
Yeah, better under linux-msm.

davidinux, 2024-11-27:
+1

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

lumag, 2024-11-27:
I started the split by preparing to remove Nexus/Pixel and WoA recipes.
Are we okay with leaving package-the-binaries recipes for HDKs and db8074 or should I move them to the external repo too?

lumag, 2024-11-27:
See #681 , #682 .

ndechesne, 2024-11-27:
thanks. Both PRs above look pretty nice cleanups. thanks.
Let's keep the HDK and original dragonboard in meta-qcom for now. they are Qualcomm reference designs, so it makes sense. If that becomes a problem in the future we can stage 'legacy' hardware in -extras.

davidinux, 2024-11-27:
-extra usually refers to new features not fully baked yet or for which there’s a lesser SLA, whilst -legacy or -obsoleted refers to leftovers that make sense to keep from an historical perspective but are no longer actively maintained nor encourages.

Might be worth considering the -legacy term for dragonboard?

D

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

lumag, 2024-11-27:
I still have hopes that one day we might have HDK binaries under the 'redistributable' licence.
DB8074 and IFC board -- probably no hopes on that side.

ndechesne, 2024-11-28:
I've given this a first try, see https://github.com/quic-yocto/meta-qcom-hwe/pull/92

My intent is to prepare a branch that is the result of the merge of the 2 unrelated git histories into a single branch. This branch *will* become meta-qcom 'master' eventually (when it's reviewed and functional). 

I've tried to explain clearly the process I followed in the PR and the commits I made

ndechesne, 2025-01-16:
I have pushed an update of the PR here: https://github.com/qualcomm-linux/meta-qcom-hwe/pull/92

I believe we are really close to be able to merge this PR. Once it's merged, we will have meta-qcom 'merged' into meta-qcom-hwe. We will run it for a couple days, and fix issues that arise, and we will then be ready to transfer meta-qcom repo completely. At which point, we will merge meta-qcom-hwe main branch into meta-qcom to finalize the transition. 

Please help review/test in the mean time!

lumag, 2025-01-24:
@ndechesne would you mind opening a PR against `meta-qcom`? I think the only remaining issue is the repolinter config.

lumag, 2025-01-24:
I think that was the plan. Or would you like to move meta-qcom first? It makes sense too...

ndechesne, 2025-01-24:
I think it's better to move first, so that we don't have 2 duplicated histories in 2 places.

lumag, 2025-01-24:
I've sent a message to @fboudra , he should be able to do the magic.

ndechesne, 2025-01-27:
This repo has just been transferred from github.com/linaro to github.com/qualcomm-linux, as agreed. 
Initial testing shows that 'things are in place', and redirection (http, git, ssh, ..) works well.

Next steps:

1. Set up Github CI for master branch using Qualcomm Runners [initiated]
2. Test Github CI for stable branches using Linaro Runners
3. Transfer relevant open issues from meta-qcom-hwe to meta-qcom
4. Transfer relevant PR from meta-qcom-hwe to meta-qcom
5. Update CODEOWNERS to restore @lumag maintainers permissions 
6. Clean up meta-qcom-hwe/main branch (remove all code and add a README)
7. Restore the mirror from git.yocto.org

If any issue is observed because of the transition, please open an issue here.

lumag, 2025-01-28:
For the second point: premerge CI is currently skipped for old branches, see https://github.com/qualcomm-linux/meta-qcom/pull/724