qualcomm-linux/meta-qcom #693: yoctoproject.org mirror no longer works

ndechesne, 2024-12-09:
meta-qcom is mirrored on git.yoctoproject.org (https://git.yoctoproject.org/meta-qcom/). It's actually a mirror set up and maintained by the Yocto people. When the mirror was set up, meta-qcom was located on github.com/ndechesne/meta-qcom, so the mirror was setup to work with this repo. 

When we migrated meta-qcom to github.com/linaro/meta-qcom, we did not update the yocto mirroring. However because github automagically redirects repo after migration, it kind of continued to work. 

However I recently started to contribute to meta-qcom again, and I made a fork of linaro/meta-qcom in my github, so ndechesne/meta-qcom was live again, and it broke the automagic redirection that github was doing. As a consequence Yocto started to mirror from ndechesne/meta-qcom instead of linaro/meta-qcom.

I talked to @halstead and he can fix the mirror, but he needs help from an admin of meta-qcom to configure the webhook in the project settings.

halstead, 2024-12-09:
I've updated the mirror scripts to point at https://github.com/Linaro/meta-qcom.git. In order to keep the repo updates fulyl current we'll need to set up a private webhook on the repo. What's the best way to communicate private information to the admins of https://github.com/Linaro/meta-qcom.git ?

lumag, 2024-12-09:
I think the best way is @fboudra , I'm not an admin of the project.

lumag, 2024-12-13:
@halstead note, we plan to change the setup and move the repo under Qualcomm organization. I'm not sure if it's worth fixing it at this point if the setup is going to change in the next few weeks.

halstead, 2024-12-13:
Okay. As long as everyone is fine with the current update timing this can
wait until after the move to Qualcomm.

ndechesne, 2025-01-27:
@halstead we've moved meta-qcom in a new Github org at https://github.com/qualcomm-linux/meta-qcom.

Can you please describe how the mirroring between git.yoctoproject.org would work for meta-qcom? e.g. explain what configuration we need to set on this repo for it to work? 

thanks

halstead, 2025-01-28:
I've updated the mirror script to point at https://github.com/qualcomm-linux/meta-qcom. In order to keep the mirror up to date with each push someone with privileges to https://github.com/qualcomm-linux/meta-qcom/settings/hooks needs to add the webhook. 

Can @fboudra or another admin email me at mhalstead@linuxfoundation.org to finish setup?

ricardosalveti, 2025-01-28:
Asking a hand from @mynameistechno here.

mynameistechno, 2025-01-29:
@halstead I'll reach out.

mynameistechno, 2025-01-29:
The webhook was already defined and recent deliveries appeared to be successful but I updated the secret per what @halstead sent, just in case he updated it. Let us know if it doesn't mirror on the next push.

ndechesne, 2025-01-30:
Thanks. everything looks to be working fine again.

closed 2025-01-30 by ndechesne