qualcomm-linux/meta-qcom #714: Find common location for machine specific partitioning related data

ricardosalveti, 2025-01-27:
Follow up from https://github.com/quic-yocto/meta-qcom-hwe/pull/67#issuecomment-2473040250, we should have a dedicated repository for the partitioning related data that is consumed by meta-qcom-hwe but not hosted by the layers.

This is so we can start sharing with other common distributions (e.g. debian) later on.

ndechesne, 2025-01-29:
After some internal discussions:

1. the .conf format was designed to support (easily) eMMC vs UFS vs NAND. eg. only contains partition information, not physical storage.  
2. however, we've agreed to store XML files instead of the .conf file format, since that's the actual file that we 'ship'
3. It is under qualcomm-linux because we intend to make releases for Qualcomm reference boards, dev boards, and we wanted some level of control on the repo. I am sure we can figure out how to open up to community platforms
4. We should have per board, not per SoC configuration files. 
5. While QCOM is unlikely to support/provide AOSP support, this can be added as a community need. Let's discuss how to do that in a proper way.

lumag, 2025-01-30:
1. The .conf file also contains storage information: ```--disk --type=ufs --size=137438953472```, ```--partition --lun=0```.
2. yay!
3. Having Bjorn and Konrad as org maintainers also provides some control on the repo. I think it makes sense to have the repo under `linux-msm/` as it provides a correct message: a shared repo with Qualcomm and community contibutions. For example, Amit Pundir might be interested in AOSP partition schemes, even for RB3gen2 or other IoT boards. We might be interested in adding partition schemes for other development kits (not limiting ourselves to just HDKs and RB boards). PostmarketOS developmers might add their recommended partition schemes, etc.
4. Agreed :-)
5. At least for some of the existing boards we were providing and generating both Linux-targeted and AOSP-targeted partitioning schemes. In some cases this was just an inertia from existing mobile XML. In other cases it helped AOSP developers. Basically for each of the boards we had either both subdirs, `aosp` and `linux`, or just one of those.

obbardc, 2025-01-30:
FWIW the --size parameter appears to be unused and I have successfully removed it from my custom rb3gen2 configs

ricardosalveti, 2025-02-04:
On a different note, we should probably align the xml versioning with the compatible boot firmware version, since we also have to align with the boot firmware partition layout expectations.

We also have the UFS provisioning xml, which is relevant (due lun sizes and configuration), but currently also managed outside the boot firmware and default partitioning data.

PR #949 (2025-07-02): Use partitions configuration from the qcom-ptool repo
We currently have two sources of partition information: qcom-ptool repo and a set of partition.conf files in meta-qcom. Remove the duplication by using the data from the qcom-ptool repo.

Closes #714
closed 2025-07-02 by lumag