qualcomm-linux/meta-qcom #717: Use firmware packages defined in linux-firmware recipe

sbanerjee-quic, 2025-01-27:
linux-firmware recipe defines quite a few firmware packages, e.g. linux-firmware-qcom-vpu.
Consider using these firmware packages for the machines defined in this layer.

lumag, 2025-02-08:
The meta-qcom-hwe [provided](https://github.com/qualcomm-linux/meta-qcom-hwe/blob/scarthgap/recipes-firmware/firmware/firmware-qcom-hlosfw_1.0.bb) firmware for the following platforms:
- qcs6490 (qcm6490, already a part of linux-firmware)
- qcs9100 (sa8775p, will be included into February release of linux-firmware)
- qcs8300 (not in linux-firmware)
- qcs615 (not in linux-firmware)

If necessary firmware for QCS615 can be easily pushed into linux-firmware, just specify whether it should be r1.0_00052.0 or r1.0_00053.0 .
For QCS8300 I'd like to understand relationship between its firmware and QCS9100 / SA8775P.

lumag, 2025-02-13:
@ricardosalveti @sbanerjee-quic after linux-firmware 20250211 do we have any missing firmware files? Can this issue be closed now?

ricardosalveti, 2025-04-08:
Let's close this one for now and reopen for the affected targets as we see the need.

closed 2025-04-08 by ricardosalveti