qualcomm-linux/meta-qcom #730: RideSX build fails during QDL

nrclark, 2025-02-03:
I'm trying to get a build up and running on my RideSX. I followed the instructions in the README, and was able to build `core-image-minimal` pretty easily (`kas` manifest used was `meta-qcom/ci/sa8775p-ride-sx.yml`). But when I go flash it using the `qdl` as per the README, I eventually get this failure:

command: `tmp/deploy/images/sa8775p-ride-sx$ qdl --debug prog_firehose_ddr.elf rawprogram*.xml patch*.xml`

Output:
```
... (omitted for length)...
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="INFO: Calling handler for configure" /></data>
LOG: INFO: Calling handler for configure
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="INFO: Storage type set to value UFS" /></data>
LOG: INFO: Storage type set to value UFS
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<response value="ACK" MemoryName="UFS" MinVersionSupported="1" Version="1" MaxPayloadSizeToTargetInBytes="1048576" MaxPayloadSizeToTargetInBytesSupported="1048576" MaxXMLSizeInBytes="4096" DateTime="Oct 17 2024 - 02:00:19" /></data>
accepted max payload size: 1048576
FIREHOSE WRITE: <?xml version="1.0"?>
<data><program SECTOR_SIZE_IN_BYTES="4096" num_partition_sectors="6" physical_partition_number="0" start_sector="0" filename="gpt_empty0.bin"/></data>

FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="INFO: Calling handler for program" /></data>
LOG: INFO: Calling handler for program
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<response value="ACK" rawmode="true" /></data>
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<response value="ACK" rawmode="false" /></data>
flashed "PrimaryGPT" successfully
unable to open zeros_33sectorS.bin...failing
user@hostname:~/work/bringup-ridesx/build/tmp/deploy/images/sa8775p-ride-sx$
```

In my image-deploy folder, I can see `zeros_1sector.bin` and `zeros_5sectors.bin` OK. But `zeros_33sectors.bin` doesn't seem to exist. Where can I find this file, or how do I build it?

ricardosalveti, 2025-02-04:
At the end of the build you should have a folder like 'core-image-base-sa8775p-ride-sx.rootfs-20250204024209.qcomflash' at tmp/deploy/images/sa8775p-ride-sx, which contains the expanded list of files with the correct names (e.g. dtb.bin, rootfs). You should be able to run qdl from this folder.

keymint.mbn is indeed missing, will check tomorrow based on downstream and validate on my hw, this is something that was pending on my list.

nrclark, 2025-02-04:
@ricardosalveti confirmed, my `deploy/images/sa8775p-ride-sx/<image>.qcomflash/` does have include most of those files, and is also missing `keymint.mbn`. Is that file something that I can build?

ricardosalveti, 2025-02-05:
@nrclark keymint.mbn is actually not required (leftover from old firmware). I created https://github.com/qualcomm-linux/meta-qcom/pull/732 and with that I'm able to flash and boot my ridesx.

With the required qualcomm account login (needed for the pdf and sail):
- https://docs.qualcomm.com/bundle/80-70015-263/resource/80-70015-263_REV_AE_Qualcomm_IQ-9_Beta_Evaluation_Kit_Quick_Start_Guide.pdf
- UFS provisioning: https://artifacts.codelinaro.org/artifactory/codelinaro-le/Qualcomm_Linux/QCS9100/provision.zip (qdl --storage ufs --debug prog_firehose_ddr.elf provision_1_2.xml)
- Flash SAIL (currently requires login/account): git clone -b r1.0_00058.0 --depth 1 https://chipmaster2.qti.qualcomm.com/home/git/qualcomm/qualcomm-linux-spf-1-0_test_device_sail.git/; cd qualcomm-linux-spf-1-0_test_device_sail/QCS9100.LE.1.0/common/build/sail_nor;  qdl --storage spinor --debug prog_firehose_ddr.elf rawprogram0.xml patch0.xml

Then flash the content from qcomflash: qdl --storage ufs --debug prog_firehose_ddr.elf rawprogram*.xml patch*.xml

```
root@qcs9100-ride-sx:~# cat /etc/os-release
ID=poky
NAME="Poky (Yocto Project Reference Distro)"
VERSION="5.1 (styhead)"
VERSION_ID=5.1
VERSION_CODENAME="styhead"
PRETTY_NAME="Poky (Yocto Project Reference Distro) 5.1 (styhead)"
CPE_NAME="cpe:/o:openembedded:poky:5.1"
root@qcs9100-ride-sx:~# uname -a
Linux qcs9100-ride-sx 6.13.0-yoctodev-standard-00087-g94ef9b00225d #1 SMP PREEMPT Sat Feb  1 14:09:41 UTC 2025 aarch64 GNU/Linux
```

ricardosalveti, 2025-02-19:
@nrclark were you able to test with the latest meta-qcom changes?

ricardosalveti, 2025-05-02:
Closing as ride-sx is currently tested as part of our normal runs, so expect to be functional.

closed 2025-05-02 by ricardosalveti