qualcomm-linux/meta-qcom #734: CI/CD lava testing for qcs9100-ride-sx

ricardosalveti, 2025-02-05:
Once https://github.com/qualcomm-linux/meta-qcom/pull/732 is merged we should be able to include qcs9100-ride-sx as part of our ci/cd lava testing, as done with rb3gen2.

* UFS provisioning: https://artifacts.codelinaro.org/artifactory/codelinaro-le/Qualcomm_Linux/QCS9100/provision.zip: qdl --storage ufs --debug prog_firehose_ddr.elf provision_1_2.xml
* Flash SAIL (currently requires login/account): git clone -b r1.0_00058.0 --depth 1 https://chipmaster2.qti.qualcomm.com/home/git/qualcomm/qualcomm-linux-spf-1-0_test_device_sail.git/; cd qualcomm-linux-spf-1-0_test_device_sail/QCS9100.LE.1.0/common/build/sail_nor; qdl --storage spinor --debug prog_firehose_ddr.elf rawprogram0.xml patch0.xml
* Then flash the content from qcomflash: qdl --storage ufs --debug prog_firehose_ddr.elf rawprogram*.xml patch*.xml

lumag, 2025-02-05:
As a note. Previously it was strongly recommended not to reprovision UFS unless absolutely required. @ndechesne might remember details.

ricardosalveti, 2025-02-05:
> As a note. Previously it was strongly recommended not to reprovision UFS unless absolutely required. [@ndechesne](https://github.com/ndechesne) might remember details.

That's right, provisioning is just required once in this case (added a note as this will probably be needed since the hw probably has old software images).

mwasilew, 2025-05-02:
RIDE-SX boot testing was added with ad4f36f9e01ba6165d366abc3c2c7bd7a74fea68.

closed 2025-05-02 by mwasilew