qualcomm-linux/meta-qcom #743: Process: broken DCO chain

lumag, 2025-02-10:
@krzk notes that recent commits have broken DCO chain, there is a DCO from the original author, but no committer's DCO.
Most likely this comes from using 'Rebase&Merge' on GitHub. In order to be consistent, we should consider fixing the way PRs are merged, so that the SoB of the committer is also recorder (or so that there is a trivial merge commit).

[image]

krzk, 2025-02-10:
Yep, this makes DCO in my opinion useless. Last 6 months of history of commits have invalid DCO. I am not even speaking of commits which lack any sort of DCO (see commit 2b3fc6461dd979b6a478a7105de50d7867c8bac8 and many more...). It's the final committer SoB which is important, because that person can certify that got the contents from author.

[image]

ricardosalveti, 2025-02-10:
> Most likely this comes from using 'Rebase&Merge' on GitHub.

Yes, that's due rebase, and it does not make it useless necessarily, but moves the verification to github in a way (since it is the one rebasing and merging, and the DCOs are verified as part of the pull-request).

The only way to have the committer there is to enable merge commits, and while not my preference, it is the only way to also keep signed commits when provided by the developer.

ricardosalveti, 2025-02-10:
> I am not even speaking of commits which lack any sort of DCO (see commit [2b3fc64](https://github.com/qualcomm-linux/meta-qcom/commit/2b3fc6461dd979b6a478a7105de50d7867c8bac8) and many more...)

Right, this was merged normally (https://github.com/qualcomm-linux/meta-qcom/pull/587), but there was no action / workflow to check DCOs there, which is not the case anymore.

krzk, 2025-02-10:
> The only way to have the committer there is to enable merge commits, and while not my preference, it is the only way to also keep signed commits when provided by the developer.

No, the only way is to use Git not Github. Using Github wrappers and then explaining - it is the only way I can do it - is rather dissapointing. Rebase with proper SoB, as Git already supports.

> and the DCOs are verified as part of the pull-request).

That's not true. The committer DCO is not verified. You mix authors SoB with DCO.

> but there was no action / workflow to check DCOs there, which is not the case anymore.

And no one reviews this code?

krzk, 2025-02-10:
> and the DCOs are verified as part of the pull-request).

As a proof that above statement is not correct, look:

[image]
Where is your certification, as the **person who put it into the Git repository** - changed or unchanged, that:

 _The contribution was provided directly to me by some other person who certified (a), (b) or (c) and I have not modified it.
I understand and agree that this project and the contribution are public and that a record of the contribution (including all personal information I submit with it, including my sign-off) is maintained indefinitely and may be redistributed consistent with this project or the open source license(s) involved._

Without your certification, the chain is gone and useless.

lumag, 2025-02-10:
I think the main issue is that somebody from outside doesn't have a way to check or verify DCO chain, because it terminates with the Author. It doesn't track the committer. On the contrary, OE-Core, meta-openembedded, Linux kernel and other projects make sure that the commit records the full DCO chain from the Author up to the final Committer. I think, the easiest way to ensure that for the meta-qcom repo would be to switch back to the 'Merge' way of merging pull requests instead of the current 'Rebase & Merge' option.

Just my additional 2c: I've always been opposing to the 'Rebase & Merge' model for the very obvious reason: it also looses orinal author's baseline. When somebody works on a feature, there is some baseline commit.  At least this way I know that the developer has checked that a particular tree worked. Then later on if the isssue arises after the PR is merged I can go back to the merged commit, try reproducing the issue there and then try reproducing the issue on the base commit. In the worst case this tells me that the developer has't actually tested something or did very light testing, etc. If the PR is rebased during the merge, I have no idea on which tree the developer has been working, what was the state, etc. The valuable information is lost.

With all of that in mind I'd kindly ask to switch the way the commits are merged in the GH for this repo.

ricardosalveti, 2025-02-10:
> I think the main issue is that somebody from outside doesn't have a way to check or verify DCO chain, because it terminates with the Author. It doesn't track the committer. On the contrary, OE-Core, [...]

Yup, not without using github, which I agree it is not ideal (ideally we should have something that can be verified with git only).

> I think, the easiest way to ensure that for the meta-qcom repo would be to switch back to the 'Merge' way of merging pull requests instead of the current 'Rebase & Merge' option.

I don't think there is any issue, at this stage, to move back to 'Merge'. We had 'Rebase & Merge' to avoid having massive branches and changes being merged at once (as meta-qcom-hwe, so we could have a more linear repository view), but now that we merged with meta-qcom, with CI/CD and an extended list of reviewers (requiring 2 approvals and so on), we should be good.

'Merge' also allows to keep gpg signed commits in place.

@ndechesne what do you think? Changing should be easy, just need to coordinate with Mark.

krzk, 2025-02-11:
I am surprise why you are defending Github workflow so much. You can achieve all this with regular Git - still have Github for CI actions and checks - but just rebase with proper DCO. You claim like there was no access to Git over Github, which is very odd - I run Github projects as well and there is *zero* problem in accessing it via Git. 

> I think the main issue is that somebody from outside doesn't have a way to check or verify DCO chain

No. There is no DCO chain here. Committer could add whatever they wanted, including code which they did not have rights to, and did not certify DCO. Also committer did not certify that actually this was received from the original author without changes.

Basically there is no certification from committer, which is the most important. There is no DCO chain here at all working and failing to understand this and keep defending this that you have some "Ci task" is again, very discouraging.

krzk, 2025-02-11:
> As a proof that above statement is not correct, look:
>
> [image] Where is your certification, as the **person who put it into the Git repository** - changed or unchanged, that:
>
> _The contribution was [...]

And to remind - this is still waiting for answer. Where is the certification of committer for this?

ndechesne, 2025-02-11:
I have no concern to enable merge commits instead of rebase. I tend to prefer this workflow for Github anyway.

@krzk @lumag can you show us an example of what you want to see for DCO for a Github based project? 

For example, the Zephyr project, which is also entirely developed on Github seems to be using a 'rebase' model for PRs, and with no committer sign off, see
https://github.com/zephyrproject-rtos/zephyr
and https://docs.zephyrproject.org/latest/contribute/guidelines.html#developer-certification-of-origin-dco

Once we converge here, we should document our DCO properly in tree.

ndechesne, 2025-02-11:
> [@ndechesne](https://github.com/ndechesne) I provided this already, assuming you follow Yocto and OE DCO v1 - I want to see the certification of every person touching the patch what DCO v1 states. I quoted even part of DCO which needs to be fulfilled/certified by person handling patch.

I am quite familiar with Yocto/OE or kernel workflow, as you know! They use an email/patch workflow which makes it very straight forward to include a signed-off tag when applying patches. It is not the case with Github, that's why I asked if there is an existing project which would do something better. The example I linked (Zephyr) is definitely a major OSS project which does not seem to be doing anything different from what we are doing here.

> 
> If you want to see the example how it is done in general, you can check any kernel commit. If you want to see how to do that on Github, well ask the provider of the forge, because I am not a Github customer service. Example for Github project while using rebase (not merges) is audioreach-topology. Probably there are many. My other Github project - neard - explicitly does not use DCO (SoBs are intentional errors) so don't take it as en example.

I don't see how DCO in meta-qcom is any different from here https://github.com/linux-msm/audioreach-topology/commit/f4ac2326d983f3b93bd46999fd9c09d9aaf4ee92. 

You've pointed a couple of commits without an author signed-off tag, and I agree that's wrong, and a bug.. but in the commit above, there is no committer signed off either. What am I missing?

krzk, 2025-02-11:
> I don't see how DCO in meta-qcom is any different from here https://github.com/linux-msm/audioreach-topology/commit/f4ac2326d983f3b93bd46999fd9c09d9aaf4ee92.
Yeah, that's also incorrect, mistake. Probably I clicked some random buttons like a monkey - I really do not know, neither care about Github.

If you go a bit further:
https://github.com/linux-msm/audioreach-topology/commit/d69032a86f7aca79a13c9109650f47bb14fccac9
https://github.com/linux-msm/audioreach-topology/commit/06ef7caa05b3e6c0d8f01d6425ba1ee5bea70406

krzk, 2025-02-11:
Anyway you are reflecting the problem how to do this on Github. I don't know. I would do this on Git, but before was said:

> Nobody has direct write access to this repository,

which is even more surprising. You cannot make a push to your own repo? You cannot write to own repo but rely 100% on Github web interface? That's the problem I would suggest to solve. Github should be here only a helper, not replacement of standard rights over repository.

ndechesne, 2025-02-11:
> Anyway you are reflecting the problem how to do this on Github. I don't know. I would do this on Git, but before was said:
>
>
> which is even more surprising. You cannot make a push to your own repo? [...]

We do not have. It's a policy we've set. We could change it of course. But it's a collaborative maintenance team. We require at least 2 maintainers to approve each PR before it can be merged. So in essence it's the development process we've decided for repos in github.com/qualcomm-linux. There is nothing 'wrong' with that. If using a merge commit can help improve the situation, then that's a much simpler change to make.

ndechesne, 2025-02-11:
> [@ndechesne](https://github.com/ndechesne) for the example of a possible practice I would point out to older Linaro-time PR merges. They have original commits, author = comitter, certified with author's DCO, then a trivial merge commit, which doesn't require a DCO as it contains no changes on its own

Both @rsalveti and myself said that we would be ok to have a merge commit. We removed it in the meta-qcom-hwe days since the development process of that tree was not 'PR based'.. So let's wait for Ricardo to chime in, and we can have a chat with the other maintainers.

krzk, 2025-02-11:
Having approved PR (which results in **0 trails in Git history** and if you move project anywhere else you loose 100% of review trails, so again you depend 100% on Github to me it is wrong, to someone else it is perfectly fine) does not mean you cannot write to repo. Unless the goal is of Github/M$ to oversee your process, so you disallow people to push but only allow Github scripts... well, then maybe Github commandline client allows it?

Or maybe this is not achievable at all in such workflow?

> For example, the Zephyr project, which is also entirely developed on Github seems to be using a 'rebase' model for PRs, and with no committer sign off, see

That's odd. To me it means DCO is just half-done and better just abandon it to make things easier. It even feels unfair - contributor is forced to signed off and certify (DCO), but maintainer does not have to. IMO, this should be opposite. Although maybe Zephyr got some guidance from Linux Foundation lawyers and it is actually fine? If so, you could use that guidance/arguments as well and close the issue/ticket.

ricardosalveti, 2025-02-11:
> That's odd. To me it means DCO is just half-done and better just abandon it to make things easier. It even feels unfair - contributor is forced to signed off and certify (DCO), but maintainer does not have to.

Isn't the case https://github.com/linux-msm/audioreach-topology/commit/f4ac2326d983f3b93bd46999fd9c09d9aaf4ee92 as well?

[image]

krzk, 2025-02-11:
@ricardosalveti I think you missed some discussion here. We already talked about this.

krzk, 2025-02-11:
I don't get what you are saying. We discussed that commit and I really do not have a clue how it is relevant here. You want to complain about audioreach, then feel free, but how is it relevant here?

> I pasted is the first one you merged from another author.

That's not true. Read again the replies. We discussed this and I provided examples of such commits.

ricardosalveti, 2025-02-11:
> > I pasted is the first one you merged from another author.
> 
> That's not true. Read again the replies. We discussed this and I provided examples of such commits.

Read again what I said as well, I'm talking what happens when you use github to perform changes, and showing what happened when the same was used in audioreach, which shows the exact same issue.

We have one basic requirement here, due the policy Nico explained, which is no direct push access, so the only way to have author owning the commit is to enable merge commits.

ricardosalveti, 2025-02-11:
> > The same happens with other commits,
> 
> Instead of accusations you could at least paste some links... Unless you mean different email? You understand that certification comes from the person?

Accusations? I'm talking about how github handles the commit owner when you perform merge via the platform, it is using the same author as it was used in https://github.com/linux-msm/audioreach-topology/commit/f4ac2326d983f3b93bd46999fd9c09d9aaf4ee92, which shows how github itself works.

krzk, 2025-02-11:
> Read again what I said as well, I'm talking what happens when you use github to perform changes, and showing what happened when the same was used in audioreach, which shows the exact same issue.

We all know how Github works and that's not the point of this discussion. We all know Github is broken. We discuss that your DCO is incomplete, to which you claimed "_, and the DCOs are verified_". We keep disagreeing on that and as argument to that you reply: I found mistake in audioreach (which we discussed already)? Yeah, sure and? DCO is still incomplete.

BTW, audioreach does not have DCO policy. I brought it as example to show you how this can look if you want to implement it, because Nico asked.

> Accusations? I'm talking about how github handles the commit owner when you perform merge via the platform, it is using the same author as it was used in https://github.com/linux-msm/audioreach-topology/commit/f4ac2326d983f3b93bd46999fd9c09d9aaf4ee92, which shows how github itself works.

Can you read previous replies?

Starting from: https://github.com/qualcomm-linux/meta-qcom/issues/743#issuecomment-2650864519
and then: https://github.com/qualcomm-linux/meta-qcom/issues/743#issuecomment-2650897754

lumag, 2025-02-11:
Can I please ask to calm down a bit. The discussion moves closer and closer to becoming a non-professional one.

ricardosalveti, 2025-02-11:
> We all know how Github works and that's not the point of this discussion. We all know Github is broken.

Github is important to this discussion as it is hosted here and due the maintenance policy we have, we need to merge changes via this platform (as Nico said, we can always discuss to change the policy, but that is a different discussion).

I already gave my +1 to enable to rebase & merge.

mynameistechno, 2025-02-12:
I disabled Rebase & Merge and left both the Merge and Squash/Merge options.

ricardosalveti, 2025-02-12:
Merge enabled and working https://github.com/qualcomm-linux/meta-qcom/pull/746.

As this is the only mechanism we have, in GitHub, to keep the original commiter when merging changes via the platform (without direct push), closing this issue.

Suggestions to the maintenance policy should be discussed in a separated issue.