qualcomm-linux/meta-qcom #782: Add a way to build all meta-qcom recipes

lumag, 2025-02-27:
agreed. we could do a bitbake world but that's rather expensive.. @lumag could we parse meta-qcom in such a way that we can create something like qcom-world virtual target?

_Originally posted by @ndechesne in https://github.com/qualcomm-linux/meta-qcom/pull/766#discussion_r1967337493_

lumag, 2025-02-27:
I'm not sure about bb internals, but something like this might work:

```
bitbake `find ../layers/meta-qcom -name "*.bb" -o -name "*.bbappend" | sed -e 's!.*/!!g' -e 's!\(_.*\)\?\.bb\(append\)\?!!g' | grep -v 'qcom-partition-confs\|firmware-qcom-boot\|qcom-gen-partition-bins\|linux-yocto\|linux-linaro-qcomlt'`
```

Fixing https://github.com/qualcomm-linux/meta-qcom/issues/713 would allow us to drop qcom-* and firmware-qcom-boot from the list of exclusions.

quaresmajose, 2025-02-27:
We can get all the recipes using the `bitbake-layers` but because there is a bug in there we need to that in 2 step:

Launch the kas container:
```bash
kas-container shell ci/qcs6490-rb3gen2-core-kit.yml
```

1. Get the layer name: 

_Inside the kas the layer is the last dir name of the path, when it should simply be qcom. I will try to propose a fix for this issue_

```bash
builder@cc4e20ef30cf:/build$ bitbake-layers show-layers 
NOTE: Starting bitbake server...
layer                 path                                                                    priority
========================================================================================================
qcom                  /build/../repo                                                          6
yocto                 /build/../work/meta-yocto/meta-poky                                     5
core                  /build/../work/oe-core/meta                                             5
```

2. Get the recipes of the `repo` layer get in [1]

```bash
builder@cc4e20ef30cf:/build$ bitbake-layers show-recipes --layer repo --recipes-only --bare
NOTE: Starting bitbake server...
Loading cache: 100% |                                                                                                               | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################################| Time: 0:00:13
Parsing of 1000 .bb files complete (0 cached, 1000 parsed). 1977 targets, 65 skipped, 0 masked, 0 errors.
=== Available recipes: ===
bootrr
debugcc
diag
esp-qcom-image
fastrpc
firmware-ath6kl
firmware-qcom-boot-dragonboard410c-sdcard
firmware-qcom-boot-qcs6490
firmware-qcom-boot-qcs9100
firmware-qcom-dragonboard-apq8060
firmware-qcom-dragonboard-apq8074
firmware-qcom-qar2130p
firmware-qcom-sm8150-hdk
firmware-qcom-sm8350-hdk
firmware-qcom-sm8450-hdk
firmware-qcom-sm8550-hdk
firmware-qcom-sm8650-hdk
hexagon-dsp-binaries
hexagon-dspso-qcom-qar2130p
hexagon-dspso-qcom-sm8150-hdk
hexagon-dspso-qcom-sm8350-hdk
hexagon-dspso-qcom-sm8450-hdk
hexagon-dspso-qcom-sm8550-hdk
hexagon-dspso-qcom-sm8650-hdk
initramfs-firmware-db8074-image
initramfs-firmware-dragonboard820c-image
initramfs-firmware-image
initramfs-firmware-mega-image
initramfs-firmware-qar2130p-image
initramfs-firmware-rb12-image
initramfs-firmware-sm8150-hdk-image
initramfs-firmware-sm8350-hdk-image
initramfs-firmware-sm8450-hdk-image
initramfs-firmware-sm8550-hdk-image
initramfs-firmware-sm8650-hdk-image
initramfs-kerneltest-full-image
initramfs-kerneltest-image
initramfs-module-copy-modules
initramfs-rootfs-image
initramfs-test-full-image
initramfs-test-image
initramfs-tiny-image
linux-linaro-qcomlt
linux-qcom-uki
mybw
packagegroup-dragonboard-apq8074
packagegroup-dragonboard410c
packagegroup-dragonboard820c
packagegroup-dragonboard845c
packagegroup-qar2130p
packagegroup-rb1
packagegroup-rb2
packagegroup-rb3gen2
packagegroup-rb5
packagegroup-sa8775p-ride
packagegroup-sm8150-hdk
packagegroup-sm8350-hdk
packagegroup-sm8450-hdk
packagegroup-sm8550-hdk
packagegroup-sm8650-hdk
pd-mapper
pil-squasher
qbootctl
qc-image-unpacker
qca-swiss-army-knife
qcom-gen-partition-bins
qcom-gen-partitions-tool
qcom-partition-confs
qcom-ptool
qdl
qmic
qrtr
qtestsign
rmtfs
rust-android-sparse
skales
tqftpserv
u-boot-qcom
```

lumag, 2025-02-27:
@quaresmajose this doesn't include recipes which we bbappend in the layer.

quaresmajose, 2025-05-02:
> [@quaresmajose](https://github.com/quaresmajose) this doesn't include recipes which we bbappend in the layer.

I need to check that