qualcomm-linux/meta-qcom #806: missing files in rawprogram0.xml for RB1

mwasilew, 2025-03-18:
When trying to use generated qcomflash tarball for RB1, the following files are not included:
 - not built: 
```
boot-erase.img
persist.img
logfs_ufs_8mb.bin
```
 - missing from tarball:
```
NON-HLOS.bin
dspso.bin
BTFM.bin
```

ndechesne, 2025-03-18:
> boot-erase.img
> persist.img
> logfs_ufs_8mb.bin

For these ones, we need to remove them from the partition conf file, they are not needed. 

> NON-HLOS.bin
> dspso.bin
> BTFM.bin

These files are in the boot firmware, they are just not deployed in the qcomflash directory. we probably want to finalize #803 before we fix it.

lumag, 2025-03-18:
I'd say, the last three files are also to be removed. It is DSP firmware (and seral other MBN/MDT files), DSP binaries and Bluetooth firmware - all the files we provide as a part of the rootfs.

ricardosalveti, 2025-03-19:
+1 to remove if not really needed.

PR #807 (2025-03-20): Fix qcomflash dir for RB1 board
As noted in #806 the qcomflash dir for the RB1 board misses several files, which can cause errors during flashing.
- Install boot image into the qcomflash dir and use it for `boot_a`/`boot_b` partitions
- Don't flash `persist` and `logfs`, they are unused by Linux
- Don't flash firmware and DSP partitions, they are unused with meta-qcom

Closes #806
commit e135d7b1 (2025-03-20): Fix qcomflash dir for RB1 board (#807)

As noted in #806 the qcomflash dir for the RB1 board misses several
files, which can cause errors during flashing.
- Install boot image into the qcomflash dir and use it for
`boot_a`/`boot_b` partitions
- Don't flash `persist` and `logfs`, they are unused by Linux
- Don't flash firmware and DSP partitions, they are unused with
meta-qcom

Closes #806
closed 2025-03-20 by ndechesne