qualcomm-linux/meta-qcom #956: Boot up issue observed with latest images on qcs9100-ride-sx, qcs6490-rb3gen2-core-kit and rb8

quic-khuzuri, 2025-07-04:
Hi,
we tried to flash the following latest images from 
https://quic-yocto-fileserver-1029608027416.us-central1.run.app/qualcomm-linux/meta-qcom/16056548144-1/poky-altcfg/qcs9075-iq-9075-evk/core-image-base-qcs9075-iq-9075-evk.rootfs-20250703171332.qcomflash.tar.gz
 
https://quic-yocto-fileserver-1029608027416.us-central1.run.app/qualcomm-linux/meta-qcom/16046639642-1/poky-altcfg/qcs6490-rb3gen2-core-kit/core-image-base-qcs6490-rb3gen2-core-kit.rootfs-20250703093017.qcomflash.tar.gz
 
https://quic-yocto-fileserver-1029608027416.us-central1.run.app/qualcomm-linux/meta-qcom/16056548144-1/poky-altcfg/qcs9100-ride-sx/core-image-base-qcs9100-ride-sx.rootfs-20250703171337.qcomflash.tar.gz

observed bootup stuck in uefi with the following Logs 

-----------------------------
Platform Init  [ 2136] BDS
UEFI Ver   : 6.0.250618.BOOT.MXF.1.0.c1-00364-KODIAKLA-1
Platform           : IOT
Subtype            : 0
Boot Device        : UFS
Chip Name          : QCS_LEMANS_INDUSTRIAL
Chip Ver           : 2.0
Chip Serial Number : 0x8AD463B0
ufs_error_log: 252
RPMB ReadCounterPkt failed, Result = 6
-----------------------------
Failed to create 'Persisted_Capsules.bin' for persisting capsules. Status: Success
Failed to create capsule raw file. Status: Out of Resources
InitStoragePointersForCapsuleRawFile returned - Out of Resources!
InitCapsuleStoragePointers returned - Out of Resources.
Failed to flush capsule raw file GUID table. Status: Invalid Parameter
LocateProtocol(DPP) returned Status:Not Found
Locate pMorPpiProtocol failed 0xE 00000010
Platform Init End : 2193
-----------------------------
[QcomBds] Removable boot path

ricardosalveti, 2025-08-04:
@quic-khuzuri can you paste the entire boot log information for at least one target? We are testing the same images in our lava instance and they are all booting correctly there.

ricardosalveti, 2025-08-04:
> Hi [@ricardosalveti](https://github.com/ricardosalveti) , the issue is observed w.r.t QualNet which seems to be resolved in the externally downloaded bins as confirmed by APT.

Oh, ok, will close the issue then, please reopen if there are issues with the images itself.

closed 2025-08-04 by ricardosalveti