qualcomm-linux/meta-qcom #1017: Supporting pushing artifacts to S3

craigez, 2025-08-15:
Based a a few recent silo'ed discussions I wanted to open an issue for a broader take on what remains to enable pushing the artifacts to S3 for the Coral GH application to facilitate testing workflows. This approach is independent of moving the workflows to AWS based runners.

We need to be able to support:
 * Internal testing - via S3 and the Coral GitHub Application
 * QC developers to browse and download the artifacts
 * LAVA testing - via an externally accessible URL (can be private or public, if private using token access)

Here are some possible steps we can take, in an incremental manner:
# Enhance the upload-private-artifact action to support both GCP and AWS:
   # There is now a branch supporting AWS S3: https://github.com/qualcomm-linux/upload-private-artifact-action/tree/aws. To make the migration easiest, it would be good to support both the GCP location and the AWS location from the main branch this action. Then we could more easily stage the transition in an incremental way.
   # Initially we would enable both locations, so we can add additional AWS based workflows, without impacting the current workflow. 
   # S3 location should be private for now, using a token and a fixed url scheme so LAVA can be supported in future
   # GCP can continue to support LAVA and for developers to download the artifacts for now
       # Eventually we can turn off the GCP storage support.
# Add contents.xml
   # My understanding is the latest preference is via support in qcom-ptool rather than direct in meta-qcom. 
   # This PR is in progress and work is being done to minimize what is needed: https://github.com/qualcomm-linux/qcom-ptool/pull/29
       # This is preferred over this approach: https://github.com/qualcomm-linux/meta-qcom/pull/992
# Coral GitHub Application can then be enabled to pick up the artifacts from S3 and copy it internally for testing workflows
   # Internal testing is now supported via S3 and Coral GH Application
# Coral GitHub Application can then support copying the artifact to a location that QC developers can retrieve it, e.g. Artifactory
    # Should be retained for at least 2 weeks, preferably 30 days
    # QC developer downloading would now be supported
# Switch LAVA to use S3 location
    # LAVA is now supported via S3
    # GCP file storage can be disabled in the upload private artifact action
# We can switch to AWS based runners anytime, independently of the above artifact workflow

There are also some internal notes here: go/qliciartifacts

craigez, 2025-08-20:
#1023

mwasilew, 2025-10-16:
Yes, this is audioreach. Here is how they prepare URLs:
https://github.com/AudioReach/meta-ar/blob/main/.github/actions/process_urls/action.yml