qualcomm-linux/meta-qcom #1029: meta-qcom CI did not notice a yocto-check-layer failure

ndechesne, 2025-08-21:
meta-qcom CI runs yocto-check-layers, and the Yocto Project CI also runs it on its Autobuilder. The Yocto CI reported a failure in meta-qcom yocto-check-layer (see https://autobuilder.yoctoproject.org/valkyrie/#/builders/85/builds/606) which our own CI did not detect. Let's investigate why.

quaresmajose, 2025-08-21:
Cloning all repos separately it doesn't fail!
```bash
git clone https://github.com/openembedded/openembedded-core
cd openembedded-core
git checkout e861561b
git clone https://github.com/openembedded/bitbake
git -C bitbake checkout 7ad404fd
git clone https://git.yoctoproject.org/meta-yocto
git -C meta-yocto checkout 85967da3
git clone https://github.com/qualcomm-linux/meta-qcom.git
git -C meta-qcom checkout 5e444cab
source oe-init-build-env
bitbake-layers add-layer ../meta-yocto/meta-poky
bitbake-layers add-layer ../meta-qcom
yocto-check-layer-wrapper ../meta-qcom --no-auto-dependency
```

quaresmajose, 2025-08-21:
> > I have replicated localy the failure on the autobuilder using poky and meta-qcom
> > git clone https://git.yoctoproject.org/poky
> > cd poky
> > git checkout a2d9103b
> > git clone https://github.com/qualcomm-linux/meta-qcom.git
> > git -C meta-qcom checkout [5e444ca](https://github.com/qualcomm-linux/meta-qcom/commit/5e444cabe7539503bdac431180be0c463f3af03f)
> > source oe-init-build-env
> > bitbake-layers add-layer ../meta-qcom
> > yocto-check-layer-wrapper ../meta-qcom --no-auto-dependency
> 
> Doing the poky test but changing the DISTRO to be the same as the one of our test also fails as well
> 
> ```
> echo 'DISTRO = "nodistro"' >> conf/local.conf
> ```
> 
> It looks like a bug on the yocto-check-layer and not in meta-qcom.

Removing also the meta-yocto-bsp layer but the behaviour is the same.

ndechesne, 2025-08-22:
ok, i nailed it down.. 

When we run yocto-check-layer with oe-core + bitbake, we end up setting DISTRO=nodistro. When DISTRO=nodistro, then neither opengl nor vulkan is set, so mesa recipe is skipped. 

```
$ bitbake-layers show-recipes mesa
NOTE: Starting bitbake server...
Loading cache: 100% |######################################################################################################################################| Time: 0:00:00Loaded 2057 entries from dependency cache.
=== Matching recipes: ===
mesa:
  meta-qcom            25.2.0-rc2 (skipped: one of 'vulkan opengl' needs to be in DISTRO_FEATURES)
  meta                 unknown (skipped: one of 'vulkan opengl' needs to be in DISTRO_FEATURES)
  meta                 unknown (skipped: one of 'vulkan opengl' needs to be in DISTRO_FEATURES)
  meta                 unknown (skipped: one of 'vulkan opengl' needs to be in DISTRO_FEATURES)
```

which is why we did not get the error. 

yocto-check-layer should be run against poky I believe. 

now it gets sublte.. since you force DISTRO=nodistro even when you cloned poky git above. However yocto-check-layer-wrapper will create a new temp build folder, and your change in local.conf is ignore.

quaresmajose, 2025-08-22:
> now it gets sublte.. since you force DISTRO=nodistro even when you cloned poky git above. However yocto-check-layer-wrapper will create a new temp build folder, and your change in local.conf is ignore.

This makes sense and that's what misled me in my conclusions. I have already confirmed and it is not possible to change the default configuration of useb by the yocto-check-layer-wrapper, which is what is used in the Autobuilder.

Since we don't use the yocto-check-layer-wrapper, I can change the configuration from DISTRO=nodistro to DISTRO=poky and we'll have the same result. I think with that DISTRO=poky change we became compliant again, should I propose this?

quaresmajose, 2025-10-22:
The check-layer tests are oficcial now running under nodistro instead of poky on the autobuilder. So we are complient again and now oficially with the OE-core nodistro.