qualcomm-linux/meta-qcom #1032: Include adbd and Startup Scripts in Ramdisk for Device Communication

smuppand, 2025-08-22:
Background
Currently, Axiom execution requires either SSH or ADB for device communication. While SSH (over WiFi or Ethernet) is commonly used, there is a risk of regression due to ongoing work to move to mainline, which may impact SSH reliability or availability.

Problem Statement
SSH over WiFi/Ethernet may not be reliable or available in all scenarios, especially with mainline kernel changes.
ADB provides a robust alternative for device communication, especially for early bring-up and automation.
At present, adbd and its startup scripts are not included in the ramdisk, limiting our ability to use ADB as a fallback or primary communication channel during early boot.

Request
Please include the following in the ramdisk

smuppand, 2025-08-25:
> [@smuppand](https://github.com/smuppand) I think the problem is with the configuration. For adb daemon to start, "/etc/usb-debugging-enabled" is expected to be present. Can you run `touch /etc/usb-debugging-enabled` then restart adbd manually and see if it works?

The device now appears in adb when using the qcom-next build and this initramfs, after implementing these suggestions. Closing the issue.
Thank you @vkraleti & @ndechesne

vkraleti, 2025-10-01:
Fixed with https://github.com/qualcomm-linux/meta-qcom/pull/1048

closed 2025-10-01 by vkraleti