qualcomm-linux/meta-qcom #1046: Enable Talos firmware CPIO in meta-qcom builds

smuppand, 2025-08-31:
The Talos platform enablement in meta-qcom is currently blocked due to the absence of the required firmware CPIO artifact.

The concerned team has confirmed that the necessary Talos firmware is now available in the upstream linux-firmware repository. Once the CPIO is integrated, we can proceed to enable Talos support in qcom-next too.

smuppand, 2025-09-01:
Okay, I'm referring to the firmware cpio.gz files not being copied to the nightly filers `https://quic-yocto-fileserver-1029608027416.us-central1.run.app/qualcomm-linux/meta-qcom/17351238185-1/qcom-distro/qcom-armv8a/`. The build artifacts are being generated and copied, but the firmware cpio.gz files are not.

lumag, 2025-09-01:
Get the firmware into Linux-firmware first. If it's not upstreamed, it doesn't exist from my POV.

closed 2026-02-23 by quic-khuzuri